Bouni/kicad-jlcpcb-tools #457: All mappings are suddenly lost

cutlasses, 2024-04-28:
**Describe the bug**
After using the plug-in and entering in many LSC values the mappings, the mappings I've entered have disappeared. If I click Mappings I can see the part mappings are there but they don't appear on the main menu, and when I generate the files the csvs are empty. I tried to import from a previously exported csv and it loaded the mappings correctly, but they still didn't appear in the main screen.

**To Reproduce**
The method to reproduce is not known to me, I just used the plug-in and the mappings were lost. MacBook M1 Pro Sonoma 14.4 

**Expected behavior**
Mappings to stay until I delete them.

# KiCad Version

```
Application: KiCad PCB Editor arm64 on arm64

Version: 8.0.1, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.13.2
	HarfBuzz 8.3.0
	FontConfig 2.15.0
	libcurl/8.4.0 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.58.0

Platform: macOS Sonoma Version 14.4 (Build 23E214), 64 bit, Little endian, wxMac

Build Info:
	Date: Mar 14 2024 13:30:33
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.84.0
	OCC: 7.7.2
	Curl: 7.87.0
	ngspice: 42
	Compiler: Clang 14.0.3 with C++ ABI 1002

Build settings:
```

zeitkunst, 2024-04-29:
I also am having this problem on an upgrade to the latest release version, using KiCAD 7.0.9. I downgraded to 2023.09 and the parts and mappings came back. I will remain on 2023.09 until this bug is fixed.

cutlasses, 2024-04-29:
> I also am having this problem on an upgrade to the latest release version, using KiCAD 7.0.9. I downgraded to 2023.09 and the parts and mappings came back. I will remain on 2023.09 until this bug is fixed.

Presumably to go back a version you need to fetch through Git directly, rather than using KiCad's "Plugin and Content Manager"?

zeitkunst, 2024-04-29:
> Presumably to go back a version you need to fetch through Git directly, rather than using KiCad's "Plugin and Content Manager"?

I first removed the new version, and then I downloaded the 2023.09 version from the Releases section here on Github.

cutlasses, 2024-04-29:
> > > I also am having this problem on an upgrade to the latest release version, using KiCAD 7.0.9. I downgraded to 2023.09 and the parts and mappings came back. I will remain on 2023.09 until this bug is fixed.
> > 
> > 
> > Presumably to go back a version you need to fetch through Git directly, rather than using KiCad's "Plugin and Content Manager"?
> 
> I first removed the new version, and then I downloaded the 2023.09 version from the Releases section here on Github, so yes, I didn't use KiCAD's Plugin and Content Manager.

Thanks!

cutlasses, 2024-04-30:
Sadly that didn't work for me. I downloaded 2023.09 but the LCSC values didn't come back.

cutlasses, 2024-04-30:
I typed all of the LCSCs in again and lost them again, so I'll wait for this bug to be fixed before using again. This time it happened when closing the PCB editor and re-opening (even though I'd clicked "Save Mappings").

Bouni, 2024-07-04:
I have a rough idea whats the problem but I'm not sure what going on.

The 3 databases (parts, totations, mappings) are stored under the plugin forlder in `jlcpcb`:

```
.\jlcpcb\
├── mappings.db
├── parts-fts5.db
└── rotations.db
```

In order to update a plugin installed via PCM you need to remove it first, which deletes the entire folder and therefore the databases as well. 

I don't know if theres a place to store data "persistent" so that it survices an update ...

cutlasses, 2024-07-05:
Is your hypothesis that the loss of mappings will only occur if you install/update a plug-in? I don't think that was the case for me.

Bouni, 2024-07-10:
@cutlasses Then I don't have an explanation what happened ...

You can export your mapping an import them, just in case this happens again.
AS I have no idea what happend, this is the only solution I can offer for the moment

cutlasses, 2024-07-20:
@Bouni could the issue be pressing the launch icon whilst the app is already running? I do all my KiCad work on my laptop with limited screen real-estate. The JLC exporter doesn't appear under the Windows list in KiCad, so I'm often losing it, and I've sometimes just pressed the launch icon again to switch to it. Could that be launching 2 instances and causing issues?

I'm annotating another board. I'm regularly saving mappings, and exporting back to the schematic. No issues so far..

cutlasses, 2024-07-21:
I entered all my LCSC parts and generated the files but noticed there was no BOM file. I tried updating the plugin and lost my values (which could be the issue you mentioned before). I'm trying to import my saved ones. It loads them to the Import menu, but they don't appear in the main list. Am I doing something wrong?

[image]

Bouni, 2024-07-25:
As far as I remember (I know this is embarrassing 😅) mappings are not loaded automatically from the mappings table!
You have to right click on a part and use the "Find LCSC from mappings" function:

[image]

You can do that for multiple parts by using the select alike function.

z2amiller, 2026-09-20:
This is a pretty old issue and the plugin has collected a lot of changes since 2024 - I'm going to close this, but if someone can provide fresh screenshots of this happening on a modern version please re-open this issue and we'll track it down - thanks!

closed 2026-09-20 by z2amiller