Bouni/kicad-jlcpcb-tools #458: Additional Maintainers

Bouni, 2024-04-29:
As everybody has noticed, I hardly have the time to maintain and enhance this plugin due to other factors takeing up my spare time.

So I really would like to have one or two co-maintainers that can help out.

@whmountains and @chmorgan are you interested?

We could setup the repo so that at least one review is required for a merge of a PR with branch protection on the main branch.

whmountains, 2024-04-30:
I'd love to be involved as well.  Thanks for the vote of confidnece!  I use this tool at least once a week for work and am able to contribute on company time as well.  My availability isn't as consistent as what @chmorgan has been demonstrating though, it all depends on what other projects we have going on at work.

I agree with @chmorgan that it would be nice if you could stick around and act as sort of a tie-breaker for decisions.  Or pick another leader.  You have brought the project this far, so I trust your judgement.

@Bouni you made a really solid foundation here.  The tool is very useful, and the codebase is organized and easy to understand.  It's been cool to see so many people coming out of the woodwork with high-quality bug reports and patches.  I look forward to seeing it maintained long-term.

whmountains, 2024-04-30:
Just wanna mention @chmorgan is definitely the more qualified contributor here, in case it's not already obvious 🤪.  If you ever need to completely hand over the reigns to someone, he should be at the top of your list.

Bouni, 2024-05-13:
@chmorgan and @whmountains I just sent you an invite. I set up branch protection and some basic rules (GH actions must pass in order to merge, ...)

Hopefully the project can get som traction with your help now!

@JLA4444 I need to look into GitHub if it is even possible to give suh fine grained access. Don't get me wrong but I don't want to many users with write access as I cannot pay really close attention at the moment  ...

chmorgan, 2024-05-14:
Thank you! Looking forward to helping out and appreciate the trust in
sharing these permissions.

whmountains, 2024-05-14:
Thank you for the confidence, @Bouni.  I'm looking forward to contributing as well.

closed 2024-05-14 by whmountains