Bouni/kicad-jlcpcb-tools #459: Hard to type in category field after searching

whmountains, 2024-04-30:
Steps to reproduce:

- Type in a search query, the slower the query the more obvious the problem will be.  `1k` is a great example.
- Click into the Category field and start typing the name of a category.  I typed "Resistors"

Expected result: I can type fluidly and the search does not re-run until I am done typing.
Actual result: There is a long lag after every keystroke while the query re-runs.

Known workaround: open the dropdown and pick your desired category from the menu.

NOTE: the problem does not occur when typing in the Subcategory field.  Hopefully it is an easy fix to apply debouncing to the Category field as well.

Bouni, 2024-07-04:
I had the idea to add an [wx.ActivityIndicator](https://docs.wxpython.org/wx.ActivityIndicator.html) but apparently the db search itself blocks the GUI refresh, so the spinner does not spin while the search is running 😑

Maybe we can utilize a thread and events in order to populate the listview after the search is done ...

chmorgan, 2025-03-29:
Typing seems like a helpful shortcut vs. scrolling through the dropdown, but when would we ever want to have a category of 'r' or 're' or 'res'? What *should* typing be used for when the category isn't a valid one? Providing a shorter dropdown as you get closer to having a match?

chmorgan, 2025-03-29:
@whmountains thoughts on this change? https://github.com/Bouni/kicad-jlcpcb-tools/pull/592 This disables typing into category and subcategory entirely (as the values are known and distinct and typing would permit users to deviate from valid categories and subcategories).

chmorgan, 2026-09-06:
Closing this one out as its been quiet for years.

closed 2026-09-06 by chmorgan