Bouni/kicad-jlcpcb-tools #467: Sorting on number in stock is LSCS assignment window not working

JLA4444, 2024-05-05:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Press assign LCSC
2. Press column to sort on stock
3. Sorting is not correct

**Expected behavior**
A clear and concise description of what you expected to happen.

# KiCad Version

```
Application: KiCad PCB Editor x86_64 on x86_64

Version: 8.0.2-unknown-202405031934~57c8fbc9fe~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.10.1
	HarfBuzz 2.6.4
	FontConfig 2.13.1
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Ubuntu 20.04.6 LTS, 64 bit, Little endian, wxGTK, X11, ubuntu, x11

Build Info:
	Date: May  3 2024 19:34:19
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.6.3
	Curl: 7.68.0
	ngspice: 42
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:

```

Bouni, 2024-07-03:
Can confirm that sorting by stock does not work.

ajf58, 2024-07-31:
I think this can be closed as fixed by https://github.com/Bouni/kicad-jlcpcb-tools/commit/42782481804cd0b89eedcd7fbbbf8283a038d6b6 , which is in release https://github.com/Bouni/kicad-jlcpcb-tools/releases/tag/2024.07.01 .

closed 2024-08-02 by Bouni