Bouni/kicad-jlcpcb-tools #471: keywords such as 5k 1% doesn't work

JLA4444, 2024-05-07:
**Describe the bug**
The % wildcard is causing issues as keywords as 5k 1% doesn't work

**To Reproduce**
Steps to reproduce the behavior:
1. Assign component
2. keywords: 5k 1%
3. You have to remove 1% for it to work

**Expected behavior**
I expect it to be able to easily find precision resistors

# KiCad Version

```
```

Bouni, 2024-07-03:
Search terms ar split by space, so we end up with 5k and 1%.
The first is ignored because its less that 3 characters wide, the later also but % needs to be escaped which I belive is nt done at the moment.
I'll look into this tomorrow (or maybe in the evening, we'll see)

chmorgan, 2024-07-31:
Tried this and seeing the same thing. Fortunately most resistor values are fractional with 3+ characters. There is likely some way we can force a db scan search on shorter tokens at the expense of performance.

Bouni, 2024-08-02:
Maybe we add a search button that can trigger a search if the keyword is to short!?