Bouni/kicad-jlcpcb-tools #472: [2024.04.03] rotation problem on SOT-23-6

lk-davidegironi, 2024-05-10:
**Describe the bug**
Rotations not working for SOT-23-6 package

**To Reproduce**
Add a Package_TO_SOT_SMD:SOT-23-6 footprint component
Check Rotations Manager: ^SOT-23-6 = 180
Generate production files
The SOT-23-6 is 90 degree rotated

**Expected behavior**
The SOT-23-6 is expected to be 180 degree rotated.

Attached the rotation file.
Sample output of CPL
U1,USBLC6-2SC6,SOT-23-6,58.928,-71.374,**90.0**,top 
Expected to be
U1,USBLC6-2SC6,SOT-23-6,58.928,-71.374,**180.0**,top 

# KiCad Version
7.0.11 on Windows 10

[rotations.csv](https://github.com/Bouni/kicad-jlcpcb-tools/files/15272762/rotations.csv)

Bouni, 2024-05-13:
I cannot reproduce this problem on my machine 

Without a rotation:

[image]

With the rotation file you attached:

[image]

I run KiCAD 8.0 but that should not make a difference ...

Bouni, 2024-05-13:
Just did the same test with 7.0.10, same result.

What rotation do you get when you remove the rotation?

lk-davidegironi, 2024-05-13:
Hello
I've updated to 2024.05.01

Something strange happens, CSV is 90 degree rotate. GUI and logger is rotated 180.

Maybe it's something related to this specific PCB?

[image]
[image]

```
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of C6 (47u/50V / CP_Elec_6.3x7.7) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of C7 (10u / CP_EIA-3216-18_Kemet-A) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of C11 (100u / CP_EIA-3528-21_Kemet-B) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of C13 (47u/50V / CP_Elec_6.3x7.7) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q1 (SS8050 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q2 (SS8050 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q3 (MMBT5401 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q4 (MMBT5401 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q5 (MMBT5401 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q6 (MMBT5401 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q7 (MMBT5401 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q8 (MMBT5401 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q9 (MMBT5401 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q10 (MMBT5401 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q11 (BSS138 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of Q12 (BSS138 / SOT-23) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U1 (USBLC6-2SC6 / SOT-23-6) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U2 (CP2102N-Axx-xQFN28 / QFN-28-1EP_5x5mm_P0.5mm_EP3.35x3.35mm) on Top Layer by 270 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U4 (XL1509-5.0 / SOIC-8_3.9x4.9mm_P1.27mm) on Top Layer by 270 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U5 (AMS1117-3.3 / SOT-223-3_TabPin2) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U6 (MCP23017_SO / SOIC-28W_7.5x17.9mm_P1.27mm) on Top Layer by 270 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U7 (LTV-247 / SOP-16_4.4x10.4mm_P1.27mm) on Top Layer by 270 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U8 (LTV-247 / SOP-16_4.4x10.4mm_P1.27mm) on Top Layer by 270 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U9 (LTV-247 / SOP-16_4.4x10.4mm_P1.27mm) on Top Layer by 270 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U10 (LTV-247 / SOP-16_4.4x10.4mm_P1.27mm) on Top Layer by 270 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U11 (MCP4725xxx-xCH / SOT-23-6) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U12 (ADS1115IDGS / TSSOP-10_3x3mm_P0.5mm) on Top Layer by 270 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U13 (MCP4725xxx-xCH / SOT-23-6) on Top Layer by 180 degrees
2024.05.13 17:13:14 - INFO - rotate -  Fixed rotation of U23 (ADS1115IDGS / TSSOP-10_3x3mm_P0.5mm) on Top Layer by 270 degrees
2024.05.13 17:13:14 - INFO - generate_cpl -  Finished generating CPL file
2024.05.13 17:13:14 - INFO - generate_bom -  Finished generating BOM file
```

slabua, 2024-10-18:
I have both SOT-23 and SOT-23-6 in the same schematic, and I am not able to set a different rotation for them separately.
In my case I need to rotate SOT-23 by 180 and SOT-23-6 by -90. Seems like the latter I adjust overrides the former. Something wrong with the regex? I also tried using the component name instead of the package, and still no luck.

slabua, 2024-10-18:
@lk-davidegironi I see you also have both SOT-23 and SOT-23-6 in the same board.
It all was fine for me until i only had one of them at once.

lk-davidegironi, 2024-10-18:
Never notice this, it maybe due to both present at same time.

Bouni, 2024-11-04:
Maybe you can set your rotation by name rather than by package? Make sure to delete the by package rotations