Bouni/kicad-jlcpcb-tools #473: i do not see any components in the JLCPCB tools list

cyberboy666, 2024-05-11:
**Describe the bug**
when i open the plugin menu i can not see any of the footprints from my circuit - there is just an empty menu. i do not know how to get the footprints from my design into here so i can start assigning part numbers to them...

[image]

(i even tried it on a new project but this didnt work either)

**To Reproduce**

- open pcbnew for any of my projects
- click on the JLCPCB tools button
- get this empty menu shown in picture above

**Expected behavior**

i would have expected all of the footprints from my circuit to be listed here so i can start assigning parts. maybe there is some step i am missing - it feels like this is a user issue i just cant figure out what to do next
# KiCad Version
7.0.9
running on pop_os installed from pop!_shop (which seems to be flatpak underneath - i tried the requests fix on readme but this doesnt seem to be a problem - requests is installed and im not getting any errors)
```
```

Bouni, 2024-05-13:
I have not looked into Flatpack for months even years and it only caused headaches in the past.
Can you install a non Flatpack version and test again?
I don't have the time at the moment to dig into Flatpack issues, sorry ...

cyberboy666, 2025-03-01:
quite delayed response - but for anyone who stumbles upon this - __it was user error!__ (and unrelated to flatpak)

- to see all my parts in the list i just needed to _select them all first_ with `CTRL + A` and then press the JLCPCB TOOLS button.

closed 2025-03-01 by cyberboy666