Bouni/kicad-jlcpcb-tools #474: Assign LCSC number function not working KICAD 8.0.2

MiguelASTavares, 2024-05-14:
**Describe the bug**
When i open the plugin non of the components on my brd appear in the window, this however can be solved if i manually select the components before opening it. Once inside the terminal window when i press the "Assign LCSC number" button no new window is created. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open the plugin
2. Select a component
3. Click in the "Assign LCSC number"
4. See error

**Expected behavior**
I was expecting the selection Window. 
# KiCad Version

```
Application: KiCad x86_64 on x86_64

Version: 8.0.2-8.0.2-0~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.10.1
	HarfBuzz 2.6.4
	FontConfig 2.13.1
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Ubuntu 20.04.6 LTS, 64 bit, Little endian, wxGTK, X11, ubuntu, x11

Build Info:
	Date: Apr 28 2024 21:38:36
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.6.3
	Curl: 7.68.0
	ngspice: 42
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
```

chmorgan, 2024-05-14:
@MiguelASTavares can you see the version of the plugin anywhere in the gui? Want to confirm its the latest version.

MiguelASTavares, 2024-05-15:
@chmorgan Can you tell me where to look for the version? In the top of the window, I have “JLCPCB Tools [dev]”. I have cloned the repository, but I have also tried to add the repo to the Plugin manager of KICAD, but had the same behavior.

bvernoux, 2024-05-20:
I had exactly same issue on Windows 11 + KiCad 8.0.2 + kicad-jlcpcb-tools 2024.05.01 until I have removed the database and let kicad-jlcpcb-tools 2024.05.01 re-creacte it from scratch

MiguelASTavares, 2024-05-20:
I have tried to downgrade the KICAD install to 8.0.1, and it worked (on windows), but in Linux I'm not seeing an easy way to downgrade. @bvernoux how did you remove the database and re-created?

bvernoux, 2024-05-21:
> I have tried to downgrade the KICAD install to 8.0.1, and it worked (on windows), but in Linux I'm not seeing an easy way to downgrade. @bvernoux how did you remove the database and re-created?

I have exit KiCad 8.0.2 (check with ProcessExplorer ... that KiCad is not still running in background and Kill it) then just remove the jlcpcb directory in the project (it include project.db ...).
Relaunch KiCad 8.0.2 and jlcpcb plugin and it will recreate jlcpcb/project.db ...

MiguelASTavares, 2024-05-21:
I tried to do this steps, but the plugin is still not working. Any other ideas?

Bouni, 2024-06-13:
@MiguelASTavares Can you give the latest release a shot?

MiguelASTavares, 2024-06-18:
I have updated the plugin it's now on: JLCPCB Tolls [2024.06.02], the behavior remains the same. I can't open the Assign LCSC number button.

Nick-Doe, 2024-06-19:
Same issue here re-installed as suggested
Started jlcpcb tools - database updated as normal
No partslist comes up
exit jlcpcb tools
restart
[image]

Exit kicad and kill remaining kicad task
Restart kicad
start jlcpcb tools
everything looks ok but cannot assign part
cannot right click paste part

Nick-Doe, 2024-06-19:
I have noticed the exception may happen several times when trying to start jlcpcb tools.  Eventually the parts display comes up and then seems to come up reliably.  Still not part assignment

Additional attempts:

Reverted back to old tools version

Install plugin via Plugin manager: [KiCAD-PCM-2024.05.01.zip](https://github.com/Bouni/kicad-jlcpcb-tools/releases/download/2024.05.01/KiCAD-PCM-2024.05.01.zip)

Overwrite mainwindow.py with changed one: https://github.com/Bouni/kicad-jlcpcb-tools/commit/30856ea190be9e6238f5f0f9df1362c782b48099

Same behavior.

Reverted back to previous Kicad   May 26 2024 00:43:30

Same behavior.

Application: KiCad PCB Editor x64 on x64

Version: 8.0.3-rc1, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW
OpenGL: Intel, Intel(R) Iris(R) Xe Graphics, 4.6.0 - Build 31.0.101.5382

Build Info:
	Date: May 26 2024 00:43:30
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.8.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1939 without C++ ABI

Nick-Doe, 2024-06-19:
With existing designs, BOM output format is now wrong.  Gerbers seem ok.  Not sure about cpl files.

[image]

Nick-Doe, 2024-06-20:
See #486

Bouni, 2024-06-21:
> With existing designs, BOM output format is now wrong. Gerbers seem ok. Not sure about cpl files. [...]

The wrong BOM whas caused bay a bug in the previous release. Please test the latest release.

Nick-Doe, 2024-06-21:
Hi Bouni,
Sorry if I wasted your time, but thanks very much for checking and getting
back.

I have installed the latest version and everything is now working fine,
part assignment and BOM output.
The behaviour is now the same on mapped and local drives.

I suspect something odd happened during the last windows update, maybe
rebooting "fixed" it.

I am seeing a delay (maybe 1-2 sec) after removing or pasting a part number
which was not there a couple of weeks ago.

Best,  Nick

chmorgan, 2025-03-29:
Closing as an update or restart fixed the issue.

closed 2025-03-29 by chmorgan