Bouni/kicad-jlcpcb-tools #475: Exception on python action plugin code

jacobw, 2024-05-17:
**Describe the bug**
"Exception on python action plugin code" window pops up when I click the plugin button in the PCB Editor. Plugin doesn't work after this point.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to PCB Editor
2. Click on JLCPCB Tools icon
3. See error
```
Traceback (most recent call last):

File "/Users/jacob/Documents/KiCad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 28, in Run
dialog = JLCPCBTools(None)

File "/Users/jacob/Documents/KiCad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 516, in __init__
self.init_store()

File "/Users/jacob/Documents/KiCad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 530, in init_store
self.store = Store(self, self.project_path, self.pcbnew.GetBoard())

File "/Users/jacob/Documents/KiCad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/store.py", line 32, in __init__
self.update_from_board()

File "/Users/jacob/Documents/KiCad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/store.py", line 232, in update_from_board
self.logger.debug(

File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/logging/__init__.py", line 1434, in debug
self._log(DEBUG, msg, args, **kwargs)

File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/logging/__init__.py", line 1589, in _log
self.handle(record)

File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/logging/__init__.py", line 1599, in handle
self.callHandlers(record)

File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/logging/__init__.py", line 1661, in callHandlers
hdlr.handle(record)

File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/logging/__init__.py", line 952, in handle
self.emit(record)

File "/Users/jacob/Documents/KiCad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 1130, in emit
wx.QueueEvent(self.event_destination, LogboxAppendEvent(

RuntimeError: wrapped C/C++ object of type JLCPCBTools has been deleted

```

I have just updated to the latest plugin release (2024.05.01)  and this started happening. The first time I click the icon it downloads the db ok, then after that this happens every time. I tried uninstalling and reinstalling and no change.

**Expected behavior**
Plugin works

# KiCad Version

```
Application: KiCad PCB Editor arm64 on arm64

Version: 8.0.2, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.13.2
	HarfBuzz 8.3.0
	FontConfig 2.15.0
	libcurl/8.4.0 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.58.0

Platform: macOS Sonoma Version 14.4.1 (Build 23E224), 64 bit, Little endian, wxMac

Build Info:
	Date: Apr 27 2024 22:24:31
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.84.0
	OCC: 7.7.2
	Curl: 7.87.0
	ngspice: 42
	Compiler: Clang 14.0.3 with C++ ABI 1002

Build settings:

```

steeley, 2024-05-19:
I  can confirm I get the same issue. A sort of workaround seems to be:
 save everything then quit and reopen Kicad.

`Application: KiCad x86_64 on x86_64

Version: 8.0.2, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.13.2
	HarfBuzz 8.3.0
	FontConfig 2.15.0
	libcurl/8.4.0 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.51.0

Platform: macOS Ventura Version 13.6.6 (Build 22G630), 64 bit, Little endian, wxMac

Build Info:
	Date: Apr 27 2024 23:01:31
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.84.0
	OCC: 7.7.2
	Curl: 7.87.0
	ngspice: 42
	Compiler: Clang 14.0.3 with C++ ABI 1002

Build settings:
`

bvernoux, 2024-05-20:
I have exactly same issue on Windows 11 + KiCad 8.0.2 + kicad-jlcpcb-tools 2024.05.01 
Exiting KiCad and relaunching it fix temporary the issue.

abregnsbo, 2024-05-20:
I also have this issue on Windows 10 + KiCad 8.0.2 + kicad-jlcpcb-tools 2024.05.01.

chmorgan, 2024-05-21:
So it looks like the issue is due to some mismatch of memory being freed on one side (python vs. qt).

There are a number of issues with similar text on stack overflow, like:

https://stackoverflow.com/questions/75531812/getting-past-runtimeerror-wrapped-c-c-object-of-type-has-been-deleted

@whmountains any thoughts on what might be going on here? Clearly its due to the message sending change but its not clear which object isn't valid or being properly retained.

chmorgan, 2024-05-21:
@Bouni any thoughts? I don't quite follow which object is deleted on the c++ side but not on the python side...

chmorgan, 2024-05-21:
or who we might be able to ask to take a look? We could revert the change until we can get a fix but the wrong thread updating the UI was causing a separate issue

DomoticX, 2024-05-22:
I also have the same error
KiCad v8.0.2 / JLCPCB Tools v2024.05.01

[image]

DomoticX, 2024-05-22:
I also have the same error on KiCad 8.0.2 and JLCPCB tools v2024.05.01
[image]

KnightHill, 2024-05-23:
Same issue as @DomoticX under Manjaro with KiCad 8.0.2. Not possible to downgrade, as many other libraries were changed during the last update.

chmorgan, 2024-05-24:
As an update I'm discussing the issue on the wxPython forums as I kind of understand the problem but not why or how it's happening in this case. Will update here when I know more.

chmorgan, 2024-05-26:
Also asked here on SO, https://stackoverflow.com/questions/78534034/wxpython-runtimeerror-wrapped-c-c-object-of-type-jlcpcbtools-has-been-delete

chmorgan, 2024-05-27:
Can someone with this issue please try #478 to see if this resolves the crash for them? With the help of Aleksandr Shvartzkop I was able to understand the issue and reproduce it here and fix it. I'm not able to reproduce the crash any more.

chmorgan, 2024-05-27:
@DomoticX can you confirm you've totally exited out of kicad and are still seeing the issue?

The crash is slightly different and I wasn't seeing it as that logging call in library.py:57 is only called when there isn't a datadir. So I modified library setup() locally to log even if it doesn't create the directory and not seeing the crash here.

There are also no other calls to addHandler() without matching removeHandler(), only remaining question is maybe quit_dialog() isn't the correct place to clean up the loggers.

DomoticX, 2024-05-27:
@chmorgan well i closed all there is and restarten KiCad, seem sto be working now withouth crashing ;-)

[image]

chmorgan, 2024-05-27:
@DomoticX whew. At least that’s consistent with the theory of operation behind the fix heh

@whmountains can you review the open PR so we can get it merged and a new version released?

Nick-Doe, 2024-06-04:
I can confirm that this fixed my issues so far
https://forum.kicad.info/t/issues-the-version-8-0-3-rc-and-jlcpcb-tools/51392

Bouni, 2024-06-07:
I just released 2024.06.01 which should include the fix for this.

closed 2024-06-07 by Bouni

bvernoux, 2024-06-07:
I confirm that issue is now fixed with 2024.06.01 (tested with Windows 11 Pro x64 + KiCad 8.0.3)