Bouni/kicad-jlcpcb-tools #481: BOM is written CPL content instead of BOM on 2024.06.02

cr1tbit, 2024-06-13:
Hi @Bouni 

You should probably revert this ASAP because it already propagated to the kicad repository.

https://github.com/Bouni/kicad-jlcpcb-tools/blob/5d497a5228437b4b703a32e1a89ba2de8f1dc3a0/fabrication.py#L294C1-L294C62

It probably wasn't intended, but if it was, jlc seems not to accept this format of BOM.

[image]

Best regards

cr1tbit, 2024-06-13:
I confirmed, [2024.06.01] works ok

Bouni, 2024-06-13:
Oh snap. I'll fix it later today 

Thanks for pointing this out.

Maybe a good time to add unit tests, just have to figure out how to do that properly

mikesmitty, 2024-06-16:
@Bouni it looks like just the Comment and Designator column headers are swapped in the BOM output, I've been swapping the header manually before uploading it and the site processes it correctly

Adesin-fr, 2024-06-19:
> @Bouni it looks like just the Comment and Designator column headers are swapped in the BOM output, I've been swapping the header manually before uploading it and the site processes it correctly

Same here ;) 
File is not recognized at JLCPCB !

Bouni, 2024-06-19:
Please test the latest release, the issue should be fixed now

PR #482 (2024-06-19): Revert changes made to BOM creation in #480
This reverts the changes made to the BOM creation in order to fix #481 

The fixed version now works for Panels as well as single PCBs

[image]

[image]
closed 2024-06-19 by Bouni