Bouni/kicad-jlcpcb-tools #483: Idea: add component count selector in "component search" section

cr1tbit, 2024-06-15:
Nuff said - Price column is a mess :) https://github.com/Bouni/kicad-jlcpcb-tools/issues/382 

I messed a bit with code, and wrote a simple PoC. I hardcoded element count to "30" for now:

[image]

Instead of hardcoding element count, an additional field could be added to the menu at the top, with target component count. Near it, a simple switch for toggling this feature should be present too.

As mentioned in linked issue, JLC API is not stable, so this feature should handle all parsing faults as gracefully as possible, and fallback to the default view on any error.

Another annoying issue I tried to deal here is "description" field, which is arguably even messier. I tried to highlight the searched phrase, to improve glance value when selecting component. 

The scenario i'm trying to deal with: looking for "100n" component with "0402" package - when I'm obviously looking for a capacitor, I get "100nH 0402" choke listed.

This challenge however defeated me - I don't know if it's even possible to print **bold** text using wx component used here (and how portable such feature would be). For now i wrap keyword in ||double pipes||, but this doesn't really look useful at all.

Here's the snippet I made - obviously requiring a lot more sanitation before being prod-ready:

```python

    self.populate_part_list(result, search_duration, self.keyword.GetValue())
    [...]

    def populate_part_list(self, parts, search_duration, keyword):
        """Populate the list with the result of the search."""

        target_count = 30
        search_duration_text = (
            f"{search_duration:.2f}s"
            if search_duration > 1
            else f"{search_duration * 1000.0:.0f}ms"
        )
        self.part_list.DeleteAllItems()
        if parts is None:
            return
        count = len(parts)
        if count >= 1000:
            self.result_count.SetLabel(
                f"{count} Results (limited) in {search_duration_text}"
            )
        else:
            self.result_count.SetLabel(f"{count} Results in {search_duration_text}")
        for p in parts:
            item = [str(c) for c in p]
            # Munge price to be more readable
            pricecol = 8 # Must match order in library.py search function
            descriptioncol = 7
            price = []
            try:
                for t in item[pricecol].split(","):
                    qty, p = t.split(":")

                    qty_low, qty_high = qty.split("-")

                    qty_low_f = float(qty_low)
                    try:
                        qty_high_f = float(qty_high)
                    except ValueError:
                        qty_high_f = float("inf")

                    if (target_count >= qty_low_f) and (target_count < qty_high_f):
                        p = float(p)
                        if p < 1.0:
                            price.append(f"{qty}: {p * 100:.2f}c")
                        else:
                            price.append(f"{qty}: ${p:.2f}")
                        self.logger.warning(f"${p:.2f}")                    

                    # if p < 1.0:
                    #     price.append(f"{qty}: {p * 100:.2f}c")
                    # else:
                    #     price.append(f"{qty}: ${p:.2f}")
                item[pricecol] = ", ".join(price)
            except ValueError:
                self.logger.warning("unable to parse price %s", item[pricecol])
            
            item[descriptioncol] = item[descriptioncol].replace(keyword, f"||{keyword}||")
            self.part_list.AppendItem(item)
```
Please tell me if it's something worth being merged-in, I could finish this and make an PR sometime in the future.

Bonus question - is it even remotely possible for kicad addon to utilize some other GUI-rendering framework? :)

Bouni, 2024-07-03:
I see your point and you are absolutely rigth (as others have been before) but wx is a bitch ...

You say you search for 100nF 0402 but your screenshot does not use the package field at all.

When I search 100nF and use the package field with 0402 I get quite sane results:

[image]

The first hit is even a basic part which is desirable in my opinion.

I have not looked into highlighting in wx listviews but its not easyly possible to hightlight a substring. maybe there's a rich text way but that means a lot of work ....

About the price thing, I kind of like your idea of having a way to define a quantity which the filters the price column and only shows the range that matches the quantity. Smart!

Bonus answer: It is, absolutely but there is now way you can install 3rd party python packages that work on all plattforms ...

I thought about building a webserverbase backend (flask, fastapi, ...) and use vuejs as a frontend, but spawning a browser window dows not work on all platfroms I guess. wx webviews a kind of stoneage, all state of the art UI frameworks need a modern JS engine.
QT is another option but is very heavy as a dependency, again, how to install that ....
So I guess we are stuck with wx 😐

cr1tbit, 2024-07-13:
I'll try to respond

1. **WX is a bitch** - yes. I spend a couple of hours trying to solve this, but the documentation is horrible, and the RTF stuff is probably windows-only.
2. **100n search** - yes, typing 100nF solve this specific case, but when you have to populate 30 different components, similar mistakes sometimes happen. The result is, you need 3x more attention in order not to make such mistakes. However, see point 1.
3. **The price thing** - I'll work on a PR for that, but can't promise I'll deliver until late august - so you can also do that (but please mention it under this issue beforehand, or anyone who'd like to do that, it's quite simple task if you know at least some python).

I'd add another submenu in the view seen on the screenshot, with amount of boards you're ordering, and toggle between all and filtered prices. In the main view, estimated total component price can be added as a bonus.

4. Alternative frontend - a lot of work required, but maybe jlc would be interested in sponsoring this work, did you ask them yet? Maybe they need a youtube video, so that their sponsorship can be mentioned :)

I'd explore a possibility of simply spawning a flask (or whatever) server, and opening the interface in browser. This would open a shitton of new possibilities! One can dream :)

cr1tbit, 2024-07-20:
Heh, even the JLC added the "highlight" feature now :)

[image]

z2amiller, 2026-09-20:
Fixed by #729

closed 2026-09-20 by z2amiller