Bouni/kicad-jlcpcb-tools #484: BOM is not generated - KiCad 8.0.3 + JLCPCB Tools 2024.06.03

TheGiolly, 2024-06-19:
**Describe the bug**
BOM is not generated and an error appears in the KiCad terminal.

**To Reproduce**
Steps to reproduce the behavior:
1. Tools
2. External Plugins
3. JLCPCB Tools
4. Generate

**Expected behavior**
BOM is generated.

# KiCad Version

```
Application: KiCad PCB Editor x64 on x64

Version: 8.0.3, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW
OpenGL: ATI Technologies Inc., Radeon RX 590 Series, 4.6.0 Compatibility Profile Context 24.3.1.240216

Build Info:
	Date: Jun  3 2024 19:04:47
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.8.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1939 without C++ ABI

Build settings:

```

# Terminal Output
```
2024.06.19 21:19:12 - INFO - generate_geber -  Successfully plotted Top layer
2024.06.19 21:19:12 - INFO - generate_geber -  Successfully plotted Silk top
2024.06.19 21:19:12 - INFO - generate_geber -  Successfully plotted Mask top
2024.06.19 21:19:12 - INFO - generate_geber -  Successfully plotted Paste top
2024.06.19 21:19:12 - INFO - generate_geber -  Successfully plotted Bottom layer
2024.06.19 21:19:12 - INFO - generate_geber -  Successfully plotted Silk top
2024.06.19 21:19:12 - INFO - generate_geber -  Successfully plotted Mask bottom
2024.06.19 21:19:13 - INFO - generate_geber -  Successfully plotted Paste bottom
2024.06.19 21:19:13 - INFO - generate_geber -  Successfully plotted Edges
2024.06.19 21:19:13 - INFO - generate_geber -  Successfully plotted V score cut
2024.06.19 21:19:13 - INFO - generate_excellon -  Finished generating Excellon files
2024.06.19 21:19:13 - INFO - zip_gerber_excellon -  Finished generating ZIP file C:/Users/preda/OneDrive/Sync/Speeduino/0.4.4d\jlcpcb\production_files\GERBER-v0.4.4d.zip
Traceback (most recent call last):
  File "C:\Users\preda\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\mainwindow.py", line 952, in generate_fabrication_data
    self.fabrication.generate_cpl()
  File "C:\Users\preda\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\fabrication.py", line 263, in generate_cpl
    if part[6] == 1: # Exclude from POS
       ~~~~^^^
TypeError: 'NoneType' object is not subscriptable
```

Bouni, 2024-06-20:
🤯 I'll try to fix this asap

Bouni, 2024-06-20:
I'm almost sure that there is no part in the database for the given and therefore this returns no result:

https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/fabrication.py#L262

Can you post a screenshot of your parts list as shown in the plugin main window?

guuuuus, 2024-06-20:
I've got the same problem, no bom is generated.
terminal output:

`Traceback (most recent call last):
  File "/home/guus/.local/share/kicad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 952, in generate_fabrication_data
    self.fabrication.generate_cpl()
  File "/home/guus/.local/share/kicad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/fabrication.py", line 263, in generate_cpl
    if part[6] == 1: # Exclude from POS
       ~~~~^^^
TypeError: 'NoneType' object is not subscriptable
`
[image]

Bouni, 2024-06-20:
Can one of you try if this fixes your issue: https://github.com/Bouni/kicad-jlcpcb-tools/pull/485

guuuuus, 2024-06-20:
Yes, you are right. 
fix is working now.
Thanks!

TheGiolly, 2024-06-20:
Can confirm that #485 fixes the issue.
Now the terminal output is the following
```
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:12 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
[... cut ...]
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - DEBUG - generate_cpl -  No matching database entry found for REF**
2024.06.20 18:44:13 - INFO - generate_cpl -  Finished generating CPL file C:/Users/preda/OneDrive/Sync/Speeduino/0.4.4d\jlcpcb\production_files\CPL-v0.4.4d.csv
2024.06.20 18:44:13 - INFO - generate_bom -  Finished generating BOM file C:/Users/preda/OneDrive/Sync/Speeduino/0.4.4d\jlcpcb\production_files\BOM-v0.4.4d.csv
```
If needed, I can send you the KiCad project for further debug.

Bouni, 2024-06-21:
Now its clear what happened!

We use a helper function to load parts into the database that ignores parts with a ref of `REF**`:

https://github.com/Bouni/kicad-jlcpcb-tools/blob/79d0d6839a2ccd8cb4c6782e420c9e82ec908947/helpers.py#L145-L151

I'll remove the debug output and merge the fix asap

PR #485 (2024-06-21): Do not fail creating BOM if part is not found in database
This is an attempt to fix #484 

I cannot reporduce this issue but I think that a part is not found in the database which results in a None returned and that leads to a failed BOM creation
closed 2024-06-21 by Bouni