Bouni/kicad-jlcpcb-tools #486: Windows Mapped Drive Problems (recent windows 11 update?)

Nick-Doe, 2024-06-20:
The problems I was having with assigning parts (issue #474) seems to be related mapped drives after a recent windows update.
I have my work files mapped to  drive I.
If I open the kicad files using 
I:\...\file.kicad_pro
then I can't assign parts or generate a correct BOM
Using C:\I\...\file.kicad_pro
everything is OK 

(assigning parts seems a lot slower?)

Bouni, 2024-06-21:
I'll see if I'm able to reproduce this issue locally

Bouni, 2024-06-21:
I cannot reproduce the behaviour you describe ...

1. I created a new project on a sahred drive (Synology NAS)
2. I Added two parts to the schematic and assigned a LCSC Number to one of them
3. I switched to the PCB editor and placed both parts
4. I opend the plugin, the alrteday assigned part shows up correct in the table
5. I assigned a number to the second part via the Assign LCSC number button
6. Again the table is correct
7. I generated the production files, BOM and CPL are both valid

Is there a difference in this workflow to whta you do?

Bouni, 2024-07-03:
I'll close this. Reopen if the problem remains

closed 2024-07-03 by Bouni

Nick-Doe, 2024-07-03:
Yes, things have been working fine