Bouni/kicad-jlcpcb-tools #488: Plugin fails on components without pads

ms270169, 2024-06-27:
I am working on Plugin Version 2024.06.04 and KiCad Version 8.0.3-rc1

I want to use component with multiple footprints, for example an Arduino Nano component consisting of two 15 pin Sockets, and get all production data automatically without manual post processing.

Because this use case is not supported by KiCad at the moment, I solve this demand in the following way:

1) Create a Arduino symbol with all 30 pins, but without LCSC field.
2) Bind that to a PCB footprint with all 30 pads on proper location.
3) Create an "empty" symbol with LCSC field for 15 pin header.
4) Creating PCB footprint for the "empty" symbol without Pads, but with F-Fab circles on proper pad location.
4) Adding the "empty" symbol two times in the schematic.
5) Locating the F-Fab circles exactly over the 30 pads of the Arduino footprint.

Inside KiCad everything works perfect. Bill of material and location data ar generated with proper values. No ERC and DRC error or warning is generated.

When starting the JLCPCB Tools plugin, everything works without warning or errors, but no BOM file is generated and the CPL file only has parts with identifiers before the "empty" symbol part.

As soon as I add one pad in the "empty" symbol footprint, the plugin works as desired, but then I get DRC errors because pad is now two times on same location. To move the pad outside of print is also not working, because the plugin is using the wrong pad location and not the correct footprint location.

I see no solution, as long as the plugin needs a footprint with pad for proper run.

ms270169, 2024-06-28:
This solution on file [fabrication.py line 102](https://github.com/Bouni/kicad-jlcpcb-tools/blob/2024.06.04/fabrication.py#L102) solves the problem.

Original version of method `get_position()`:
```py
def get_position(self, footprint):
        """Calculate position based on center of bounding box."""
        pads = footprint.Pads()
        bbox = pads[0].GetBoundingBox()
        for pad in pads:
            bbox.Merge(pad.GetBoundingBox())
        return bbox.GetCenter()
```
Problem fixed with:
```py
 def get_position(self, footprint):
        """Calculate position based on center of bounding box."""
        try:
            pads = footprint.Pads()
            bbox = pads[0].GetBoundingBox()
            for pad in pads:
                bbox.Merge(pad.GetBoundingBox())
            #self.logger.info(" => %s", bbox.GetCenter())
            return bbox.GetCenter()
        except:
            self.logger.info("WARNING footprint %s: original position used", footprint.GetReference())
            return footprint.GetPosition()
``

Bouni, 2024-07-03:
Can you provide a screenshot? I don't get what exactly you try to achieve.
Also, a broad except is not a good thing to do, can you let me know what exception you get for your part so that we can handle this properly?

ms270169, 2024-07-04:
You get an "Index out of bound" Exception because in [fabrication.py#L105](https://github.com/Bouni/kicad-jlcpcb-tools/blob/2024.06.04/fabrication.py#L105) you access `pads[0]`, but if there are no pads you get an exception. Because you never catch any exceptions in your plugin, the script is aborted without any error dialog to the user.
I would also recommend to add an exception catch at the plugin start to ensure an error dialog if anything goes wrong.

The basic "routing" problem (multiple footprints for one component which is not supported by KiCad) is described at the beginning of this issue. I look for any solution in the KiCad community, get some hints but no good solution. The procedure I described above works fine, and it is not a big concern to add a try/catch on proper location in plugin.

klaus-liebler, 2024-07-05:
Same problem in my use case: I use pad-less footprints to place silkscreen drawings (polarity hint, warning symbols...) on my pcb. The project is here: https://github.com/klaus-liebler/labathome/tree/master/labathome_pcb15

Regards, Klaus

closed 2024-07-19 by Bouni