Bouni/kicad-jlcpcb-tools #490: "Do not place" attribute ignored when creating BOM file

ms270169, 2024-07-01:
I am working on Plugin Version 2024.06.04 and KiCad Version 8.0.3-rc1

When creating the production files, I was wondering, that JLCPCB *Bill of Material* list includes components which are marked in KiCad with the attribute *"Do not place"* (in german version *"Nicht bestücken"*).

That causes the need to manually exclude these parts. With the following change in [fabrication.py method generate_bom()](https://github.com/Bouni/kicad-jlcpcb-tools/blob/2024.06.04/fabrication.py#L284) it is possible to solve this problem.

Original version:
```py
def generate_bom(self):
        """Generate BOM file."""
        bomname = f"BOM-{Path(self.filename).stem}.csv"
        add_without_lcsc = self.parent.settings.get("gerber", {}).get(
            "lcsc_bom_cpl", True
        )
        with open(
            os.path.join(self.outputdir, bomname), "w", newline="", encoding="utf-8"
        ) as csvfile:
            writer = csv.writer(csvfile, delimiter=",")
            writer.writerow(["Comment", "Designator", "Footprint", "LCSC"])
            for part in self.parent.store.read_bom_parts():
                if not add_without_lcsc and not part[3]:
                    continue
                writer.writerow(part)
        self.logger.info("Finished generating BOM file %s", os.path.join(self.outputdir, bomname))
```
Improved version:
```py
def generate_bom(self):
        """Generate BOM file."""
        bomname = f"BOM-{Path(self.filename).stem}.csv"
        add_without_lcsc = self.parent.settings.get("gerber", {}).get(
            "lcsc_bom_cpl", True
        )
        with open(
            os.path.join(self.outputdir, bomname), "w", newline="", encoding="utf-8"
        ) as csvfile:
            writer = csv.writer(csvfile, delimiter=",")
            writer.writerow(["Comment", "Designator", "Footprint", "LCSC"])
            for part in self.parent.store.read_bom_parts():
                components = part[1].split(",")
                for component in components:
                    for fp in self.board.Footprints():
                        if fp.GetReference() == component and fp.IsDNP():
                            components.remove(component)
                            part[1] = ','.join(components)
                            self.logger.info("Component %s has 'Do not placed' enabled: removing from BOM", component)
                if not add_without_lcsc and not part[3]:
                    continue
                writer.writerow(part)
        self.logger.info("Finished generating BOM file %s", os.path.join(self.outputdir, bomname))
```

The "Do not place" parts are now not in the BOM-file, but they remain in the CPL file. That causes the following warning message when the files are imported in JLCPCB web-page:
```
Error

The below parts won't be assembled due to data missing.
R23,R25,R28,R1,R6,R9,R31,C3,J6,R10 designators don't exist in the BOM file.
```
I think it is a good idea to let popup this warning, because it indicates that some parts are not ordered and placed, and you can see which parts are concerned. If everything is ok, you can close the message dialog and proceed the order flow...

Bouni, 2024-07-03:
KiCAD has seperate flags for "do not populate" and "exclude from BOM"

[image]

These flags should be used if one does not want a part in the BOM or the CPL.

Doing assumptions will lead to someone else show up in the issues and wanting to have another behaviour.

I hope I get your description right, if not, let me know

ms270169, 2024-07-04:
At first the table columns order should be changed. I am working on an Ubuntu system and the table shows the columns *Reference*, *Value* and *Footprint*. If I expand the dialog window manually (fast way clicking on title bar does not work) to complete witdh I see also the columns *LCSC* and *Type*, but not the columns *BOM* and *POS*. To see them I have to scroll right, but then I don't see the *Reference* column. Uaaaa...
I would recommend to locate the columns BOM and POS at third location (after *Value*).

Regarding your comment. I marked components which should neither be purchased nor equipped by PLCPCB with "Nicht bestücken" (do not place), which is readable by script with the method `isDNP()`. 

To my surprise, I found these parts in the CPL and BOM file, which meant that JLCPCB had the parts in the shopping list and the placement list. And all the time manually to exclude these parts is not very comfortable.

At the moment I am not sure about the right procedure to handle components which should not be handled by JLCPCB. I think these parts should be in the BOM, because I will buy them later elsewher and solder them by myself. At the moment I use the BOM flag which is also not a perfect solution, because these parts are marked with a big red X in schematic. Later other people will be confused about this, because these components will be available on the print. And it conflicts with spare parts which should not be placed on final print version...

Maybe an additional config file in the production_files folder would be the proper clean solution. But that needs a bigger change in the plugin script.

z2amiller, 2026-04-10:
Fixed by: #723, #722, #489