Bouni/kicad-jlcpcb-tools #492: Update of parts DB fails

Bouni, 2024-07-03:
I guess we reached the limits of GH actions runner disk space 😬

https://github.com/Bouni/kicad-jlcpcb-tools/actions/runs/9772117526/job/26975987280#step:6:519

`sqlite3.OperationalError: database or disk is full`

So, what can we do about this?

1. Let this run on another server and fetch the database from there. Quick fix
2. Refactor and build a API as discussed in #450 . Maybe the best way, but a lot of work

Bouni, 2024-07-03:
It seems that @yaqwsx 's jlcparts has the same issue, an that's the source for our parts db ...

https://github.com/yaqwsx/jlcparts/issues/132

ukd1, 2024-07-12:
Fixed by #498

Bouni, 2024-07-19:
@ukd1 

Unfortuantely this still fails with no space left on the device ...

You can see it in the annotations here: https://github.com/Bouni/kicad-jlcpcb-tools/actions/runs/10005331382

Bouni, 2024-07-26:
I just searched for a solution for the disk space problem and stumbled upon this post: https://github.com/actions/runner-images/issues/2840#issuecomment-790492173

I gave it a try and it did the trick, the database files are built again successfully 🥳

closed 2024-07-30 by Bouni