Bouni/kicad-jlcpcb-tools #497: No files are generated in the latest nightly 8.99.0-unknown-5c22162d85

kholia, 2024-07-10:
**Describe the bug**
When the generate fabrication files button is clicked, nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
- Click generate fabrication files
- Nothing happens

**Expected behavior**
Gerbers, BOM and POS should be generated.

**KiCad Version**

```
Application: KiCad Schematic Editor x86_64 on x86_64

Version: 8.99.0-unknown-5c22162d85~178~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.17

Platform: Ubuntu 22.04.4 LTS, 64 bit, Little endian, wxGTK, X11, xubuntu, x11
OpenGL: AMD, RENOIR (renoir, LLVM 15.0.7, DRM 3.54, 6.5.0-41-generic), 4.6 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2

Build Info:
	Date: Jul  8 2024 21:19:40
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.81.0
	ngspice: 42
	Compiler: GCC 11.4.0 with C++ ABI 1016

Locale: 
	Lang: en_US
	Enc: UTF-8
	Num: 1,234.5
```

**Debug log**

```
Traceback (most recent call last):
  File "/home/user/.local/share/kicad/8.99/3rdparty/plugins/kicad-jlcpcb-tools/mainwindow.py", line 951, in generate_fabrication_data
    self.fabrication.generate_geber(layer_count)
  File "/home/user/.local/share/kicad/8.99/3rdparty/plugins/kicad-jlcpcb-tools/fabrication.py", line 141, in generate_geber
    popt.SetPlotViaOnMaskLayer(False)  # Set this to True if you need untented vias
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetPlotViaOnMaskLayer'. Did you mean: 'GetLegacyPlotViaOnMaskLayer'?
Traceback (most recent call last):
  File "/home/user/.local/share/kicad/8.99/3rdparty/plugins/kicad-jlcpcb-tools/mainwindow.py", line 951, in generate_fabrication_data
    self.fabrication.generate_geber(layer_count)
  File "/home/user/.local/share/kicad/8.99/3rdparty/plugins/kicad-jlcpcb-tools/fabrication.py", line 141, in generate_geber
    popt.SetPlotViaOnMaskLayer(False)  # Set this to True if you need untented vias
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetPlotViaOnMaskLayer'. Did you mean: 'GetLegacyPlotViaOnMaskLayer'?
Traceback (most recent call last):
  File "/home/user/.local/share/kicad/8.99/3rdparty/plugins/kicad-jlcpcb-tools/mainwindow.py", line 951, in generate_fabrication_data
    self.fabrication.generate_geber(layer_count)
  File "/home/user/.local/share/kicad/8.99/3rdparty/plugins/kicad-jlcpcb-tools/fabrication.py", line 141, in generate_geber
    popt.SetPlotViaOnMaskLayer(False)  # Set this to True if you need untented vias
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetPlotViaOnMaskLayer'. Did you mean: 'GetLegacyPlotViaOnMaskLayer'?
Traceback (most recent call last):
  File "/home/user/.local/share/kicad/8.99/3rdparty/plugins/kicad-jlcpcb-tools/mainwindow.py", line 951, in generate_fabrication_data
    self.fabrication.generate_geber(layer_count)
  File "/home/user/.local/share/kicad/8.99/3rdparty/plugins/kicad-jlcpcb-tools/fabrication.py", line 141, in generate_geber
    popt.SetPlotViaOnMaskLayer(False)  # Set this to True if you need untented vias
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetPlotViaOnMaskLayer'. Did you mean: 'GetLegacyPlotViaOnMaskLayer'?
```

kholia, 2024-07-10:
As a temporary workaround, I used the following hacky patch:

```diff
diff --git a/fabrication.py b/fabrication.py
index b98440e..7f11a7e 100644
--- a/fabrication.py
+++ b/fabrication.py
@@ -138,15 +138,10 @@ class Fabrication:
 
         popt.SetSubtractMaskFromSilk(True)
 
-        popt.SetPlotViaOnMaskLayer(False)  # Set this to True if you need untented vias
+        popt.GetLegacyPlotViaOnMaskLayer()
 
         popt.SetUseAuxOrigin(True)
 
-        # Tented vias or not, selcted by user in settings
-        popt.SetPlotViaOnMaskLayer(
-            not self.parent.settings.get("gerber", {}).get("tented_vias", True)
-        )
-
         popt.SetUseGerberX2format(True)
 
         popt.SetIncludeGerberNetlistInfo(True)
```

Bouni, 2024-07-10:
I can reproduce the error, but neet to figure out how to control if vias are tented or not via the API ....

NikolayMurha, 2024-07-16:
> As a temporary workaround, I used the following hacky patch:
>
> ```diff
> diff --git a/fabrication.py b/fabrication.py
> index b98440e..7f11a7e 100644
> --- a/fabrication.py
> +++ b/fabrication.py
> @@ -138,15 [...]

I try to use tis solution and i have GERBER zip archive geerated. But I havent BOM file and I have empty CPL file :(
No errors in console:

```
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Top layer
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Silk top
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Mask top
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Paste top
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Inner layer 1
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Inner layer 2
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Bottom layer
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Silk top
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Mask bottom
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Paste bottom
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted Edges
2024.07.16 19:27:52 - INFO - generate_geber -  Successfully plotted V score cut
2024.07.16 19:27:52 - INFO - generate_excellon -  Finished generating Excellon files
2024.07.16 19:27:52 - INFO - zip_gerber_excellon -  Finished generating ZIP file /Users/nikolay/Projects/LegitForce/cm4-carrier-board/jlcpcb/production_files/GERBER-cm4-carrier-board.zip

```

Bouni, 2024-07-19:
Fixed in #502

closed 2024-07-19 by Bouni