Bouni/kicad-jlcpcb-tools #500: Help understanding the rotation feature

FrightRisk, 2024-07-14:
I am on the 2024.07.01 version. When I change a rotation, how do I save it to the database? Matt's data is in CSV format. Your program uses a database that is not text readable. How do I back it up so that the "update" function or something else doesn't obliterate my rotations? And if you do update the database so that I want changes you have on github or wherever you are getting them from, how can I merge that? For example, your database (or do you just get matt's CSV and import it?) might have some components I don't know need rotating and I should get those. But I have modified the database with components of my own. If I update, I would have to have a list somewhere to manually re-enter all those rotations. Maybe this is what the import and export buttons can help with? Thank you! Amazing tool.

Bouni, 2024-07-30:
- You can export your current rotations via the export button into a csv
- When you import rotations from a csv, all entries that match will be overwritten
- The update button downloads matt's csv and imports it, which will overwrite all matching entries
- You can export to csv, update and re-import you csv in order to keep your settings, if you just want to keep a sub set, you have to edit the csv first in Excel

FrightRisk, 2024-07-30:
Phenomenal. Thank you.

closed 2024-08-02 by Bouni