Bouni/kicad-jlcpcb-tools #501: Some JLC part numbers missing in the

FrightRisk, 2024-07-14:
**Describe the bug**
Some parts seem to be missing in the list when I open the tool. I can't see a pattern

**To Reproduce**
Steps to reproduce the behavior:
1. Open the JLC Tool
2. Scroll through the list
3. Some parts are listed, but the part numbers are missing. The parts DO wind up in the BOM and POS files

**Expected behavior**
Expect to see all parts have a number where they are entered in KiCad in a "JLC Part #" field

# KiCad Version

```
Application: KiCad PCB Editor x64 on x64

Version: 8.0.2, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Apr 27 2024 23:05:17
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:

```
C201,202, and 202 do not show the LCSC Part number of C2972719 yet they show in the BOM the tool creates

[image]

Bouni, 2024-07-19:
Thats strange, I can't reproduce this issue here:

[image]

How did you set the LCSC value? In the schematic or via the plugin?

FrightRisk, 2024-07-19:
In the schematic. Puzzles me too. The text appears in both the schematic and PCB editor:

[image]

gyohng, 2024-07-20:
@FrightRisk
Could you attach a minimal project where this can be reproduced? Also, can you quickly check if my fork behaves differently, or if it has the same problem?
https://github.com/gyohng/kicad-jlcpcb-tools

MichaelRColton, 2024-10-15:
I commented on #543 that I've been having an issue that sounds similar to this one (not sure if the two issues are related?)

MichaelRColton, 2024-10-15:
https://github.com/user-attachments/assets/1020aae6-84cc-43d8-a708-23cd1a6d6ba2

All I'm doing here is repeatedly opening and closing the tool. Hopefully this is a helpful clue? Anything else I can provide?

Bouni, 2024-10-16:
@MichaelRColton Thats very strange, thank yu for that video. That maybe help figuring out whats going on!

tanaes, 2025-05-19:
Just made a similar comment in #543 about some similar behavior, but here's what helped for me:

1) Clicked "Settings" in the plugin window, and ensured that "LCSC numbers from schematic have priority"

and

2) Checked my LSCS numbers for trailing whitespace. There's often a trailing space when I copy them into the schematic BOM, and this seems to break the automatic lookup.

I'd be curious to see if this helps in your case too!

z2amiller, 2026-09-26:
This is a pretty old bug and I can't repro, so I'm going to close it for now.
If anyone has a fresh repro for this issue, could you attach it here?
All of the videos/screenshots here have expired :-(

z2amiller, 2026-09-26:
Oh weird, now all of the links render for me.  It looks like possibly some kind of data race?  There's been a lot of work in this area
since KiCad 8 - if anyone has a repro for this please let me know and I'll try to dig into it, but I'm not able to repro myself.
(mac or crossover)

closed 2026-09-26 by z2amiller