Bouni/kicad-jlcpcb-tools #504: Idea: add component grouping in main view

cr1tbit, 2024-07-21:
[image]

I find myself always clicking "select alike parts" during JLC component selection. I understand that there are cases where you want to manually replace a single component here, but the defaut mode is choosing the same LCSC number for all of them.

There should be another "option" on the right bar, that will allow grouping all matching components.

Another argument - due to WX limitation, after coming back from component selection submenu, the list is always fully redrawn - because of that, the view will always go to the top. With large BOM, this becomes a bit annoying quite fast. Grouping components would not fix this directly, but the list will become a lot shorter, so the scrollback issue will not be that bad then.

chmorgan, 2024-07-21:
Yeah the idea of 'auto select alike parts' would be a cool feature to add, and/or grouping by alike parts, otherwise its hard to see how many 'alike' parts that were selected.

Bouni, 2024-07-23:
You're are totally right, I want to refactor the component list for a long time but have no idea how to achieve this.

PRs are welcome 😉