Bouni/kicad-jlcpcb-tools #507: Error after downloading DB

CoffeeTableShoe, 2024-07-28:
After installing the plugin the download of the DB started and an error occured. No parts are visible in the tool now.
This is after a fresh install and no other plugins installed.

**To Reproduce**
Steps to reproduce the behavior:
1. Open plugin
2. Wait for the download and extraction to finish.
3. See error

**Log**
```
2024.07.28 17:14:53 - DEBUG - download -  Removing any spurious old zip part files...
2024.07.28 17:14:55 - DEBUG - download -  Download parts db chunk 1 with a size of 76.29MB
2024.07.28 17:15:14 - DEBUG - download -  Download parts db chunk 2 with a size of 76.29MB
2024.07.28 17:15:33 - DEBUG - download -  Download parts db chunk 3 with a size of 76.29MB
2024.07.28 17:15:52 - DEBUG - download -  Download parts db chunk 4 with a size of 76.29MB
2024.07.28 17:16:12 - DEBUG - download -  Download parts db chunk 5 with a size of 76.29MB
2024.07.28 17:16:34 - DEBUG - download -  Download parts db chunk 6 with a size of 76.29MB
2024.07.28 17:16:51 - DEBUG - download -  Download parts db chunk 7 with a size of 76.29MB
2024.07.28 17:17:13 - DEBUG - download -  Download parts db chunk 8 with a size of 76.29MB
2024.07.28 17:17:31 - DEBUG - download -  Download parts db chunk 9 with a size of 76.29MB
2024.07.28 17:17:54 - DEBUG - download -  Download parts db chunk 10 with a size of 76.29MB
2024.07.28 17:18:16 - DEBUG - download -  Download parts db chunk 11 with a size of 76.29MB
2024.07.28 17:18:38 - DEBUG - download -  Download parts db chunk 12 with a size of 76.29MB
2024.07.28 17:18:57 - DEBUG - download -  Download parts db chunk 13 with a size of 76.29MB
2024.07.28 17:19:20 - DEBUG - download -  Download parts db chunk 14 with a size of 38.06MB
2024.07.28 17:19:32 - DEBUG - download -  Combining and extracting zip part files...
Traceback (most recent call last):
  File "C:\Users\alex_\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\mainwindow.py", line 571, in populate_footprint_list
    if not self.store:
           ^^^^^^^^^^
AttributeError: 'JLCPCBTools' object has no attribute 'store'. Did you mean: 'Restore'?
```

# KiCad Version
```
Application: KiCad Schematic Editor x64 on x64

Version: 8.0.4, release build

Libraries:
	wxWidgets 3.2.5
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW
OpenGL: NVIDIA Corporation, NVIDIA GeForce GTX 1660 SUPER/PCIe/SSE2, 4.6.0 NVIDIA 536.23

Build Info:
	Date: Jul 17 2024 03:26:47
	wxWidgets: 3.2.5 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.8.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1939 without C++ ABI

Build settings:

```

Bouni, 2024-07-30:
This is not the first time somebody reports this but I absolutely cannot reproduce it.
The error says `'JLCPCBTools' object has no attribute 'store'.` but self.store is in the `__init__` method of `JLCPCBTools` 🤷🏽‍♂️

Bouni, 2024-07-30:
How did you install the plugin?

b73f9, 2024-08-04:
Hey, have the same problem - installed via the kicad plugin manager / adding a new repo method from the README. 
If it helps: 
[image]
is the KICAD version and the JLCPCB tools plugin is reporting as being `2024.07.02` in the titlebar.

b73f9, 2024-08-04:
Ah, I think I know what it might be related to - I tried running it on an empty board initially (with just some edge connector pads, but no actual placeable components - if i run it with anything to actually place, it works fine)

Bouni, 2024-08-05:
@CoffeeTableShoe did you the same "mistake"?

toastedcrumpets, 2024-10-18:
> Ah, I think I know what it might be related to - I tried running it on an empty board initially (with just some edge connector pads, but no actual placeable components - if i run it with anything to actually place, it works fine).
> 
> So seems harmless / not a terribly big impact for all practical purposes

I confirm this issue arises on an empty board using current release (2024.10.01) and Kicad 8.0.6.
Can we add a guard somewhere to skip this?

chmorgan, 2026-09-06:
I think this is resolved. Closing it out due to inactivity.

closed 2026-09-06 by chmorgan