Bouni/kicad-jlcpcb-tools #518: Improve part list

Bouni, 2024-08-06:
I wanted for a long time to replace the part list with something that lets me add, change or delete entries without having to re-populate the entire list every time a change is made.

I think the way to go is using a `DataViewCtrl` in combination with a `PyDataViewModel`.

That also allows to sort the columns using anatural sort algorithm without having the database to do the sorting.

Bouni, 2024-08-06:
I'm currently working on a POC but that takes some time as it is a massive change

Bouni, 2024-08-06:
For those who are interested, I created a small test wx frame that uses `DataViewCtrl` and a data model `PyDataViewModel`.

DataViewCtrl: https://gist.github.com/Bouni/28651a1a6cc2dd4d2749612097801f97
datamodel: https://gist.github.com/Bouni/659b6f8a10132b6536a010f83f785c02

[image]

Natural Sort works without database, adding elements, modifiying elements and deleting elements works as well.

Still to do: Select alike, center icons

chmorgan, 2024-08-06:
@Bouni what would the purpose be to add/update/remove entries or is that just showing the low level functionality of not having to rebuild the list each time?

Bouni, 2024-08-07:
Yes, Just test if everything we need works. With buttons for this test but later  automatically

Bouni, 2024-08-07:
Ok, so I added the datamodel and the DataViewCtrl into the Plugin as a basic test and it seems to work fine!

[image]

All Sorting is done without the database, I only poulate the List initialy from the database

Bouni, 2024-08-07:
I work on a PR that integrate this into the plugin, the changes will quite extensive as a lot of logic is no longer necessary

closed 2024-08-08 by Bouni