Bouni/kicad-jlcpcb-tools #524: Empty BOM file

ozitronics, 2024-08-09:
**Describe the bug**
The generated BOM file is empty - just contains column headers. 

**To Reproduce**
Steps to reproduce the behavior:
1. Assign LCSC number to each component
2. Click on 'Generate'
3. Open ".csv" BOM file
4. File is empty except for column headers

**Expected behavior**
BOM file containing LCSC part numbers as per JLCPCB requirements

# KiCad Version

```
```
Application: KiCad x86_64 on x86_64

Version: 7.0.11-7.0.11~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.18

Platform: Linux Mint 21.3, 64 bit, Little endian, wxGTK, cinnamon, x11

Build Info:
	Date: Feb 21 2024 19:16:04
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 11.4.0 with C++ ABI 1016

Build settings:
	KICAD_SPICE=ON

Bouni, 2024-08-09:
Unfortunately I'm on vacation until the 20th so I cannot really look into it.
I guess it's a v7 issue, you can try if it happens also with v8. You can install them side by side. Make sure to backup your project, once opened in v8 you can no longer open it in v7

chriscla, 2024-08-12:
Check if the parts have the attribute "Exclude from Bill of Materials" set

block3r-1, 2024-08-14:
I have the same problem running KiCad 7.0.8 under Windows 10 - on my main computer I have an older version of this addon (but the same version of KiCad) and everything is working fine. Now, switched to another computer, installed newest addon version from repo and the BOM file is empty. Will check if the 8 works different.

block3r-1, 2024-08-14:
I have the same problem running KiCad 7.0.8 under Windows 10 - on my main computer I have an older version of this addon (but the same version of KiCad) and everything is working fine. Now, switched to another computer, installed newest addon version from repo and the BOM file is empty - in the log at the bottom of the window there is no "generate_bom" command visible.

Will check if the 8 works different, for now just to have work done I have installed 2024.04.03 version and BOM generation works fine.

buganini, 2024-08-19:
I had the same issue with KiCad 7 and got solved with KiCad 8

lk-davidegironi, 2024-09-19:
Possibly solved here: https://github.com/Bouni/kicad-jlcpcb-tools/issues/535#issuecomment-2360655030
At least for me. Soon I'll post a merge request.

commit 321dad59 (2024-09-20): fix #524 empty BOM file on KiCad 7 (#541)

* fix #524 empty BOM file on KiCad 7

* fix ruff issues

---------