Bouni/kicad-jlcpcb-tools #525: Feature Request: Export Additional Layers

lilindian16, 2024-08-15:
I am in the process of developing flex PCBs. JLC requires a dedicated layer for outlining the Stainless Steel stiffener with the specs included in the zipped gerber. My current work-around is to draw the details on the User.Comments layer and then edit _kicad-jlcpcb-tools/fabircation.py_ by adding ("Stiffener", Cmts_User, "Stainless Steel Stiffener") to the plot_plan_top list in the _generate_geber(self, layer_count=None)_ function. This exports the User.Comments layer as the Stiffener layer.

It would be good to have the option to include additional layers for fabrication file generation in this toolchain. That way we can keep a dedicated and named layer in the editor that we can also easily export - something like a tick-box with the default exported layers already selected. Kicad includes this feature in the built-in gerber exporter

 [image]

Bouni, 2024-08-23:
I see your point but strugle to see how we could easily include that so that the normal user does not get affected.
I don't want to replicate the exporter settings of KiCAD as the whole point of the plugin is that you can press export and get what JLC needs.

Maybe we could add a setting "export User layers" which then exports all user layers into the gebrer zip.
For sure there are to many in there for your use case but you could still delete the not need ones from the zip.

Do you have a better idea than that?

zevv, 2024-12-20:
A viable alternative would be to have a whitelist of 'special' layers that will be exported if these exist, or export all layers that have a specific prefix or suffix (for example `JLC_`): one could then name a layer "JLC_stiffener_BOT_0.2_mm_PI" to have it exported.

Bouni, 2025-03-25:
I will think about a way to realize this. Renaming layers seems a bit inconvenient. 
Maybe a new Setting that lets the user choos which layers should be exported and a button to reset to default!?

zevv, 2025-03-25:
> I will think about a way to realize this. Renaming layers seems a bit inconvenient. Maybe a new Setting that lets the user choos which layers should be exported and a button to reset to default!?

Will these settings be persistent so they will be stored as part of the project? It would be nice if the right layers are still exported if a colleague orders a new batch next year :)

lilindian16, 2025-03-25:
I made a pull request recently (https://github.com/Bouni/kicad-jlcpcb-tools/pull/581) with an approach to achieve this using RDBs suggestion of creating a “whitelist”. The implementation makes the production tool export any additional user defined layers that have JLC_ prefix in the name of the layer. Since the prefix creates a whitelist, you don’t need to have a setting item to enable the feature and the additional layers are exported automatically if present

z2amiller, 2026-04-10:
Fixed by: #581

closed 2026-04-10 by z2amiller