Bouni/kicad-jlcpcb-tools #530: Crash with 'no such table: meta' (trying to show last update in title bar) when opening KiCad JLCPCB Tools

willfurnell, 2024-09-06:
**Describe the bug**
When opening the KiCad JLCPCB Tools the following traceback appears:

[image]

I believe this is probably a regression introduced by this commit: https://github.com/Bouni/kicad-jlcpcb-tools/commit/3bc706cf2998a211321ac7e358b952888d65407d - not too familiar with the code of this library, but from a quick look from what I can gather it's probably trying to get the last update from the database before the database has even been created, as there's not a 'jlcpcb' folder in my project at this stage of launching the plugin. And even with a database that was there, the meta table hadn't been created yet as it was from an older version of the plugin.

**To Reproduce**
Steps to reproduce the behavior:
Grab the latest commit from GitHub
Open KiCad
Open any board
Click on the JLCPCB Tools icon

**Expected behavior**
The plugin loads the interface

# KiCad Version

Also happens on 8.99 from today.
```
Application: KiCad x64 on x64

Version: 8.0.4, release build

Libraries:
	wxWidgets 3.2.5
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jul 17 2024 03:26:47
	wxWidgets: 3.2.5 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.8.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1939 without C++ ABI

Build settings:

```

rohmer-brightwell, 2024-09-11:
Same problem here.

JoHo94, 2024-09-15:
Same for me. Could not install it via the KiCAD PCM because it did not show up, so i installed it via git, but abouth error showes up.

Bouni, 2024-09-16:
Maybe when there is no database yet, we can't read the last_update 🤔 
I'll try to fix this asap!

PR #534 (2024-09-16): Fix #530
Fix #530 If theres no database yet we can't fetch data from it.
commit b0cef604 (2024-09-16): Fix #530 (#534)
closed 2024-09-16 by Bouni