Bouni/kicad-jlcpcb-tools #531: Crash after trying to import rotations file

FrightRisk, 2024-09-09:
**Describe the bug**
I can't use the plugin anymore. When I created a new project, I tried to import the rotations file from another project. After that I see this:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the rotations option
2. Add an exported CSV file with corrected rotations
3. Leave KiKad and try to open it again.

**Expected behavior**
I expected to be able to load the plugin after that change.

# KiCad Version

```
Application: KiCad PCB Editor x64 on x64

Version: 8.0.4, release build

Libraries:
	wxWidgets 3.2.5
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jul 17 2024 03:26:47
	wxWidgets: 3.2.5 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.8.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1939 without C++ ABI

Build settings:

```

FrightRisk, 2024-09-09:
Answered myself. The .CSV file had entries in the wrong format. There is no error checking, so it corrupted the database. Had to uninstall and re-install.

Bouni, 2024-09-11:
Sorry to hear that, I put it on my todo list but cannot say when I have time to look into this ...

chmorgan, 2025-03-29:
@FrightRisk can you provide a project and csv file that would let us reproduce? If so I'll fix it.

closed 2026-09-07 by z2amiller