Bouni/kicad-jlcpcb-tools #532: Please can you make the bom and designators match the format that JLCPCB specifies?

adrianblakey, 2024-09-10:
**Describe the bug**
Add the plugin to KiCad 8
Run it on a PCB
Examine the bom and designators files and compare to the formats that JLCPCB require: 
https://jlcpcb.com/help/article/How-to-generate-the-BOM-and-Centroid-file-from-KiCAD
https://jlcpcb.com/help/article/bill-of-materials-for-pcb-assembly

Tye column order and naming differs from the order and naming emitted by the plugin. It's a pain to have to edit the files everytime and reorder them :-)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
csv files are produced according to the JLCPCB specification.

# KiCad Version

```
Application: KiCad x86_64 on x86_64
Version: 8.0.4, release build
Libraries:
    wxWidgets 3.2.6
    FreeType 2.13.3
    HarfBuzz 9.0.0
    FontConfig 2.15.0
    libcurl/8.9.1 OpenSSL/3.3.2 zlib/1.3.1 brotli/1.1.0 zstd/1.5.6 libidn2/2.3.7 libpsl/0.21.5 libssh2/1.11.0 nghttp2/1.63.0 nghttp3/1.5.0
Platform: Arch Linux, 64 bit, Little endian, wxGTK, Wayland, gnome, wayland
OpenGL: Intel, Mesa Intel(R) UHD Graphics (TGL GT1), 4.6 (Compatibility Profile) Mesa 24.2.2-arch1.1
Build Info:
    Date: Jul 18 2024 08:25:54
    wxWidgets: 3.2.5 (wchar_t,wx containers) GTK+ 3.24
    Boost: 1.83.0
    OCC: 7.7.2
    Curl: 8.8.0
    ngspice: 42
    Compiler: GCC 14.1.1 with C++ ABI 1019
Build settings:
```

Bouni, 2024-09-17:
Are you sure you generated the BOM ans CPL with this plugin?

This is a BOM from the plugin:

[image]

And this the CPL:

[image]

The only difference I can spot is that the Plugin names the last column of the BOM "LCSC" instead of "LCSC Part Number" but the header doesn't matter, only the order of the columns.

Bouni, 2024-10-11:
I'll close this. Re-open if the problem is still around

closed 2024-10-11 by Bouni