Bouni/kicad-jlcpcb-tools #535: 2024.09.01 plugin doesn't show up with macOS 13.6.9 anymore

Cocoader, 2024-09-16:
**Describe the bug**
I updated the plugin from 2024.07.02 to 2024.09.01 and it vanished from the External Plugins dropdown menu. Removing and reinstalling 2024.09.01  has the same effect.

**To Reproduce**
Steps to reproduce the behavior:
1. Updating from 2024.07.02 to 2024.09.01 via PCM or installing 2024.09.01 via PCM
2. Plugin disappears from External Plugins dropdown menu

**Expected behavior**
Launching the plugin.

# KiCad Version
Application: KiCad PCB Editor x86_64 on x86_64

Version: 8.0.5, release build

Libraries:
	wxWidgets 3.2.5
	FreeType 2.13.2
	HarfBuzz 8.3.0
	FontConfig 2.15.0
	libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.51.0

Platform: macOS Ventura Version 13.6.9 (Build 22G830), 64 bit, Little endian, wxMac
OpenGL: ATI Technologies Inc., AMD Radeon Pro 5500M OpenGL Engine, 2.1 ATI-4.14.1

Build Info:
	Date: Sep  6 2024 19:30:43
	wxWidgets: 3.2.5 (wchar_t,wx containers)
	Boost: 1.84.0
	OCC: 7.7.2
	Curl: 7.87.0
	ngspice: 42
	Compiler: Clang 14.0.3 with C++ ABI 1002

Build settings:

Bouni, 2024-09-17:
I just tested on Windows 11 with KiCAD 8.0.5 and have no issues.

So this might be a MacOS only issue.

Can you please go to Preferences -> PCB Editor -> Action plugins

[image]

Down next to the button you should see an icon with an attention mark.

Click it an post a screenshot of the error message.

Cocoader, 2024-09-17:
You were right about the exclamation mark. This is the output.

`  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Users/daniel/Documents/KiCad/8.0/3rdparty//plugins/com_github_bouni_kicad-jlcpcb-tools/__init__.py", line 3, in <module>
    from .plugin import JLCPCBPlugin
  File "/Users/daniel/Documents/KiCad/8.0/3rdparty//plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 7, in <module>
    from .mainwindow import JLCPCBTools
  File "/Users/daniel/Documents/KiCad/8.0/3rdparty//plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 39, in <module>
    from .partdetails import PartDetailsDialog
  File "/Users/daniel/Documents/KiCad/8.0/3rdparty//plugins/com_github_bouni_kicad-jlcpcb-tools/partdetails.py", line 12, in <module>
    from .lcsc_api import LCSC_API
  File "/Users/daniel/Documents/KiCad/8.0/3rdparty//plugins/com_github_bouni_kicad-jlcpcb-tools/lcsc_api.py", line 9, in <module>
    class LCSC_API:
  File "/Users/daniel/Documents/KiCad/8.0/3rdparty//plugins/com_github_bouni_kicad-jlcpcb-tools/lcsc_api.py", line 34, in LCSC_API
    def download_bitmap(self, url: str) -> io.BytesIO | None:
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
`

Cocoader, 2024-09-17:
Unfortunately this will not be possible because like in windows, macOS bundles it's own - I agree - ancient version of python 3.9.13. This isn't my system python version but it will only use:
`/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/`

I opened up an issue with the KiCad devs some time ago https://gitlab.com/kicad/code/kicad/-/issues/18586#note_2074090846 because I came across this issue with multiple plugins already. This isn't an easy task to standardise on the KiCad side and I think for now macOS users have a better chance to talk with the plugin devs directly and hope they still consider supporting python 3.9 (according to the KiCad devs some older still from KiCad supported linux distros also use 3.9)

ubidefeo, 2024-09-18:
@Cocoader 
do you think there's a way to use `pyenv` to trick KiCAD to use another version?
I jumped on the update this morning and now lost the ability to generate production files 🥲 
Maybe I should revert to the older version of this plugin

ubidefeo, 2024-09-18:
Never mind, I have edited the `| None` in two points and now the plugin works
I'm sure there are some edge cases which will break the plugin, but for now it's more important that I get my work done :D 

`lcsc_api.py` line 34
```
def download_bitmap(self, url: str) -> io.BytesIO:
```

`store.py` line 134
```
def set_stock(self, ref: str, stock: int):
```

Cocoader, 2024-09-18:
@ubidefeo 
>do you think there's a way to use pyenv to trick KiCAD to use another version?

I'm more of a python user - I can read it mostly but don't know the breaking changes that come with minor version increases - than a developer and only use pyenv locally not in KiCad. I did try it in the past but couldn't make it work. At some point Apples Gatekeeper / Notarization kicks in, which can be disabled of course, but it seemed too difficult for me. My reasoning was if it was that easy KiCad would already have standardised it on all platforms and wouldn't stick with 3.9 on macOS - which the KiCad devs kind of confirmed in the following issue. I did open an issue with the KiCad devs https://gitlab.com/kicad/code/kicad/-/issues/18586#note_2074090846 but I think the macOS user base is just too small - I didn't see any related issues or even comments on this issue beside the KiCad devs. The only thing that kind of works without losing Notarization is installing missing packages like 'numpy' inside the application folder with pip directly.

>Never mind, I have edited the | None in two points and now the plugin works
I'm sure there are some edge cases which will break the plugin, but for now it's more important that I get my work done :D

>lcsc_api.py line 34
`def download_bitmap(self, url: str) -> io.BytesIO:`
 
>store.py line 134
`def set_stock(self, ref: str, stock: int):`

I like your approach. Maybe @Bouni could have a look at that and evaluate what changes between `2024.07.02` and `2024.09.01` broke compatibility with python 3.9 - maybe it's just that?

ubidefeo, 2024-09-18:
@Cocoader 

I did go and check the issue, but I couldn't put myself through getting a GitLab account just to comment on it.

There sure are advantages in using fallback to `None` on returned types, too bad it breaks the plugin on Mac.
I just started using KiCAD 3 weeks ago to redesign a board I initially made in Eagle.

Finding this plugin has been a godsend, and today I was able to generate a BOM+CPL which just worked on JLCPCB which is thrilling.

@Bouni , any chance you can add a `try/except` at `__init__` time to distinguish between systems that will accept the conditional type and act accordingly?

I know we're a minority, but our heart is in a good place 🤣

Cocoader, 2024-09-18:
@ubidefeo I understand that because I also only created an account to comment on the GitLab Repo of KiCad.
I also transitioned from EAGLE to KiCad some time ago because Fusion360 Electronics is just not useable.

@Bouni Could using unions instead of the | operator be a compromise to keep the compatibility with python 3.9 on macOS?
https://docs.python.org/3.10/library/typing.html#typing.Union 
What do you think about #540?

lk-davidegironi, 2024-09-19:
Same problem for me on Windows 10, KiCAD 7.0.11, shipped with python 3.9.16
While I have no problem on the same system with KiCAD 8.0.5

@ubidefeo mod fix the problem

Waiting for a wide-band python 3.9 - 3.11 solution

Cocoader, 2024-09-19:
@lk-davidegironi This is good to know it isn't just a macOS thing. Does #540 fixes this for your KiCAD 7.0.11 on Windows 10 too?

lk-davidegironi, 2024-09-19:
~ Update ~ plugin is working, db has been downloaded, but on KiCAD 7.0.11 I have a empty BOM file generated, like the issue here: https://github.com/Bouni/kicad-jlcpcb-tools/issues/524

lk-davidegironi, 2024-09-19:
@Cocoader it fixed the issue but still I have empty BOM. Do you too have empty BOM?

Cocoader, 2024-09-19:
@lk-davidegironi I think our problems are related but not exactly the same - some do overlap. 

1. _KiCad 8 with Windows10_ works as is because it is bundled with **Python 3.11.5** which accepts the new syntax for unions which were introduced in Python 3.10 and doesn't have the problem in #524.
2.  _KiCad 8 with macOS_ **doesn't work** because it is bundled with **Python 3.9.13** and **doesn't accept** the new syntax for unions which were introduced in Python 3.10. This will only need #540.
3. _KiCad 7 with Windows10_ **doesn't work** because it is bundled with **Python 3.9.16** and **doesn't accept** the new syntax for unions which were introduced in Python 3.10 **and** something related to #524. This will need #524  **and** #540.
4. I didn't test other platforms with KiCad 7.

So for KiCad 8 compatibility (python3.9 - 3.11+) on all platforms we will need #540 and for KiCad 7 compatibility (python3.9) on all platforms at least #524  **and** #540 - maybe there are still issues after this.

There is also the question which KiCad versions https://github.com/Bouni/kicad-jlcpcb-tools/blob/088c804e6dae3279db99fa769aa2a2ad82e2b918/README.md?plain=1#L195 @Bouni wants to support in the future.

Bouni, 2024-09-20:
Hey guys, 

thank you all for your efforts, I'll replace the `|` in th etype hint with a union as suggested (@Cocoader s PR).
However, this will get a problem by itself in the near future as python will deprectae the `typing` module soon.
Maybe there will be a possible workaround, we'll see.

The KiCAD variation on different ecosystems is a huge headache to me, especially because I can only test on Windows and Linux ...

Bouni, 2024-09-20:
I just read the posts over on KiCADs GitLab (https://gitlab.com/kicad/code/kicad/-/issues/18586#note_2074090846) and it is even worse than I thought. Linux uses the systems python version which logically can be everything ...
I always test on Arch linux which as a rolling release distro hast always the most recent versio.

Maintaining a functioning plugin is very hard 😑

closed 2024-09-20 by Bouni

Cocoader, 2024-09-20:
@Bouni Thank you for the swift implementation and the new release. 

>The KiCAD variation on different ecosystems is a huge headache to me, especially because I can only test on Windows and Linux ...

>Maintaining a functioning plugin is very hard 😑

I fully agree with you. At that time I opened this [issue](https://gitlab.com/kicad/code/kicad/-/issues/18586#note_2074090846) with the KiCad devs I could only assess the plugin situation from a macOS / windows user perspective. I thought it would be a isolated macOS issue with a small user base but unfortunately it became apparent it does effect every platform one way or another.

Maybe you can give your opinion to this issue as a plugin developer to the KiCad devs because it will impact almost all plugin users. KiCad with unreliable plugin support depending on your platform would be quite a setback.

Bouni, 2024-09-20:
@chmorgan thanks for your offer, I'll @ you if I need OSX help