Bouni/kicad-jlcpcb-tools #543: After assign LCSC number table not updating

DmitryBorisenko33, 2024-09-29:
https://github.com/user-attachments/assets/1fd4f93a-9b7d-420e-a436-0a628ba43980

whmountains, 2024-10-09:
Hi @DmitryBorisenko33, I'm wondering if there was an exception behind the scenes.  Can you try it again while opening the scripting console and going to the "Dispatch" tab?  (Scripting console should be in the "tools" menu, not sure what it would be in your language.)

If there is an exception while the scripting console is open, KiCAD should pop up a window showing the error message.  At least, that is what has worked for me.

Bouni, 2024-10-11:
I can't reproduce this issue here. What version of Windows do you run?

[image]

MichaelRColton, 2024-10-15:
I've been having intermittent issues too, where when I leave the Tools window and come back (or quit KiCAD and come back) some of my LCSC numbers have disappeared. Also, sometimes if that happens and I close the Tools window and reopen it some (all?) come back. Not sure if this is the same issue, but maybe related?

M-Wehning, 2024-12-11:
This issue occurs for me too. The table only updates after closing and reopening the plugin. If I select multiple parts to edit, it only updates the one selected first. I am running KiCAD 9.4.2, Windows 11 23H2, as well as todays version of the JLCPCB tools plugin

tanaes, 2025-05-19:
I have been having similar issues, and I found that two things helped:

1) Clicked "Settings" in the plugin window, and ensured that "LCSC numbers from schematic have priority"

and

2) Checked my LSCS numbers for trailing whitespace. There's often a trailing space when I copy them into the schematic BOM, and this seems to break the automatic lookup. @Bouni it might be worth adding a whitespace `strip()` to this bit?