Bouni/kicad-jlcpcb-tools #546: All LCSC numbers appear on silkscreen

Carbon225, 2024-10-04:
I don't know if this is a feature but after assigning part numbers they all show up on the silkscreen which looks like this:

[image]

[image]

I tried searching online how to bulk hide these fields but found no way (other that scripting).

Using Edit -> Edit Text & Graphics Properties does not work on these fields.
If I select ALL filters in that menu and set Visible -> False, all graphics disappear except these LCSC numbers.

Is this something that everyone deals with or am I the only one with this problem?

I'm using kicad 8.0.5 on ubuntu 22.

luftaquila, 2024-10-06:
I'm new to this plugin and suffering from the same problem. Is this behavior new that checking out to an older release will fix this?

Carbon225, 2024-10-07:
@luftaquila you are right, I just tried with version 2024.07.02 and those numbers do not get added to the silkscreen.

My steps:

1.  `cd /path/to/plugins`
2. `git clone https://github.com/Bouni/kicad-jlcpcb-tools.git --branch 2024.07.02 --depth 1`
3. Create new project
4. Add RP2040, R and C
5. Switch to board view and assign part numbers with the plugin
6. Click "Export to schematic"
7. Reload the project
8. Still no numbers on the silkscreen

After assigning part numbers with the most recent version, the numbers are immediately visible on the silkscreen.
With the old version the numbers do not appear. When "Exporting to schematic" the LCSC field is present on each footprint but it has Show = False.
I think the new version sets by default Show = True.

Carbon225, 2024-10-07:
So I fixed it with this horrible thing in `helpers.py`

```python
134 def set_lcsc_value(fp, lcsc: str):
135     """Set an lcsc number to the first matching propertie of the footprint, use LCSC as property name if not found."""
136     lcsc_field = "LCSC"
137     for field in fp.GetFields():
138         if re.match(r"lcsc|jlc", field.GetName(), re.IGNORECASE) and re.match(
139             r"^C\d+$", field.GetText()
140         ):
141             lcsc_field = field.GetName()
142     fp.SetField(lcsc_field, lcsc)
           # Added this:
143     for field in fp.GetFields():
144         if re.match(r"lcsc|jlc", field.GetName(), re.IGNORECASE) and re.match(
145             r"^C\d+$", field.GetText()
146         ):
147             field.SetVisible(False)
```

BenRoe, 2024-10-07:
I have the same problem. Kicad 8.05 and latest Plugin version.
I am on Mac 15.0

whmountains, 2024-10-09:
I did a bit of testing.  This issue does not happen when the LCSC number is set from the schematic, something I often do.  But these fields are appearing for all components which I select after the fact using the GUI of this plugin.

whmountains, 2024-10-09:
PR submitted to fix this issue based on your snippet, @Carbon225.  New LCSC fields will be marked as hidden but existing fields will not be touched.

If you have a messed-up schematic and don't want to manually hide the LCSC field on every component, try running the following snippet at the scripting console:

```
import pcbnew

board = pcbnew.GetBoard()

for fp in board.GetFootprints():
    field = fp.GetFieldByName("LCSC")
    if field:
        field.SetVisible(False)
        
pcbnew.Refresh()
```

You will need to copy-paste one line at a time.  Pasting the whole snippet doesn't seem to work.  Press enter twice after `field.SetVisible(False)` to get back to the top level and run the refresh command.

BenRoe, 2024-10-10:
@whmountains thanks for the PR I haven't tested it yet, but you snippet works.
It's also important, copy paste the part below without indent and line by line.

```
field = fp.GetFieldByName("LCSC")
if field:
field.SetVisible(False)
```

closed 2024-10-11 by Bouni

nanokarlo, 2024-10-15:
KiCad 8.0 i have the problem  
[image]

whmountains, 2024-10-15:
@Bouni hasn't released a new version yet.  See my last comment and the clarification from BenRoe for a one-time fix.  You could also try downgrading to 2024.07.02.

Bouni, 2024-10-16:
I'll release a new version in the next few hours. It's still 6am at the moment, so be patient please 😅

nanokarlo, 2024-10-16:
@Bouni All ok take your time, 6.am on my side too XD

PR #547 (2024-10-11): Fix LCSC numbers appearing on silkscreen
Fixes #546

The Python scripting API is a bit awkward.  It would be nice if `pcbnew.FOOTPRINT.SetField` returned the resulting field so we could call `SetVisible(False)` on it.

I looked for ways to bypass the call to `GetFieldByName()`.  The [source code for SetField](https://docs.kicad.org/doxygen-python-8.0/pcbnew_8py_source.html#l19337) has some ideas.  But it wasn't working quite right and I would have to modify the function signature to take `pcbnew` as an argument for use in standalone mode.  I think we would also have to modify our standalone im [... cut]