Bouni/kicad-jlcpcb-tools #550: Loading parts DB fails

dhoepfl, 2024-10-23:
**Describe the bug**
Loading the parts DB fails with an exception “… library.py, line 368, … no such column: parts”.

I think the problem is introduced by commit ed495c0. The SQL statement is changed from

```
query = '''SELECT "LCSC Part" AS lcsc, "Stock" AS stock, "Library Type" AS type FROM parts WHERE parts MATCH ?'''
for number in lcsc:
    match_query = f'"LCSC Part:{number}"'
    cur.execute(query, (match_query,))
```

to

```
query = """SELECT "LCSC Part" AS lcsc, "Stock" AS stock, "Library Type" AS type FROM parts WHERE parts MATCH :number"""
cur.execute(query, {"number": number})
```

This drops the `LCSC Part` column name.

I think, this should be

```
query = """SELECT "LCSC Part" AS lcsc, "Stock" AS stock, "Library Type" AS type FROM parts WHERE "LCSC Part" MATCH :number"""
```

**To Reproduce**
Steps to reproduce the behavior:
1. Open JLCPCB-Tools

**Expected behavior**
The main window should open.

# KiCad Version
KiCad 8.0.5
JLCPBC tools 2024.10.01

travisjayday, 2024-11-03:
Can confirm this problem exists. Ran into it on Kicad 8.0 plugin version 2024.10.01, made the change as described by OP, and the window would succesfully open again.

dhoepfl, 2024-11-05:
Problem is solved (mostly).

What I did:

* Uninstalled plugin
* Applied pending uninstall
* Checked: No longer there
* Installed version 2024-11-01
* Applied pending
* Open PCB
* Open plugin (unrelated error, might be a bug in KiCad)

<details>
<summary>Error log</summary>

```
--- Logging error ---
Traceback (most recent call last):
  File "C:\Program Files\KiCad\8.0\bin\Lib\logging\__init__.py", line 1113, in emit
    stream.write(msg + self.terminator)
    ^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'write'
Call stack:
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\plugin.py", line 28, in Run
    dialog = JLCPCBTools(None)
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\mainwindow.py", line 478, in __init__
    self.init_library()
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\mainwindow.py", line 497, in init_library
    self.library = Library(self)
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\library.py", line 48, in __init__
    self.setup()
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\library.py", line 54, in setup
    self.logger.info(
Message: "Data directory 'jlcpcb' does not exist and will be created."
Arguments: ()
2024.11.05 11:24:14 - INFO - setup -  Data directory 'jlcpcb' does not exist and will be created.
2024.11.05 11:24:14 - INFO - setup -  Data directory 'jlcpcb' does not exist and will be created.
2024.11.05 11:24:14 - INFO - setup -  Data directory 'jlcpcb' does not exist and will be created.
--- Logging error ---
Traceback (most recent call last):
  File "C:\Program Files\KiCad\8.0\bin\Lib\logging\__init__.py", line 1113, in emit
    stream.write(msg + self.terminator)
    ^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'write'
Call stack:
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\plugin.py", line 28, in Run
    dialog = JLCPCBTools(None)
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\mainwindow.py", line 478, in __init__
    self.init_library()
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\mainwindow.py", line 497, in init_library
    self.library = Library(self)
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\library.py", line 49, in __init__
    self.check_library()
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\library.py", line 72, in check_library
    self.create_rotation_table()
  File "C:\Users\xxx\Documents\KiCad\8.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\library.py", line 235, in create_rotation_table
    self.logger.debug("Create SQLite table for rotations")
Message: 'Create SQLite table for rotations'
Arguments: ()
2024.11.05 11:24:14 - DEBUG - create_rotation_table -  Create SQLite table for rotations
2024.11.05 11:24:14 - DEBUG - create_rotation_table -  Create SQLite table for rotations
2024.11.05 11:24:14 - DEBUG - create_rotation_table -  Create SQLite table for rotations
```
</details>

* **List is empty**
* Parts DB is downloaded
* **List is (still) empty**
* Close plugin
* Reopen plugin
* **List is populated**

So the plugin is usable now but there should be a refresh after downloading the parts DB (and the UI should be disabled while downloading the DB).

Bouni, 2024-11-07:
I noticed that the list is not populated after a DB download but I had no time to look into it so far ...

Good that it works for you otherwise

chmorgan, 2025-03-30:
#593 should fix the db reloading after download issue, @Bouni, @dhoepfl, would appreciate a review / test to confirm

dhoepfl, 2025-03-31:
> [#593](https://github.com/Bouni/kicad-jlcpcb-tools/pull/593) should fix the db reloading after download issue, [@Bouni](https://github.com/Bouni), [@dhoepfl](https://github.com/dhoepfl), would appreciate a review / test to confirm

Works for me:

* Deleted parts db
* Downloaded it using the old plugin version (confirmed the problem still shows)
* Deleted parts db
* Downloaded parts db using the plugin version from main

List updates.
(I was confused for a moment because that only happens after closing the "download succeeded" alert but functionally it is fine).

closed 2025-06-10 by Bouni