Bouni/kicad-jlcpcb-tools #552: Add the thumbs up option for extended parts.

dynfer, 2024-11-11:
Would it be possible to add the thumbs up option next to extended parts that JLC doesn't charge extra?

Bouni, 2024-12-16:
What do you mean by "thumbs up option"?

dynfer, 2024-12-16:
JLC has an option when picking parts on the BOM section that certain Extended parts have a Thumbs up mark next to the extended text. This means that even if the part is extended no additional fee is accounted for it basically making it a basic part.

Bouni, 2024-12-16:
Can you share a screenshot?

FDSoftware, 2024-12-17:
> What do you mean by "thumbs up option"?

these are "extended but preferred" parts:
From what I understood from support, they are extended parts that are already available on some machine, so they do not charge you the fee.
ex:
https://jlcpcb.com/partdetail/Sunlord-SDFL1608Q4R7KTF/C1034

> Preferred Extended Parts, they will not be charged Feeders Loading fee for Economic PCB assembly.

or on the UI:
https://jlcpcb.com/parts/basic_parts
[image]

ZzEeKkAa, 2025-05-26:
+1 for the request

dynfer, 2025-06-10:
> Unfortunately this flag is not part of te data returned by the API endpoint we currently use.
> 
> I figured out that the website is doing POST requests against `https://jlcpcb.com/api/overseas-pcb-order/v1/shoppingCart/smtGood/selectSmtComponentList/v2` with a bunch of json data, for example
> 
> {
>     "currentPage": 1,
>     "pageSize": 25,
>     "presaleType": "stock",
>     "searchType": 2,
>     "keyword": "SDFL1608Q4R7KTF",
>     "componentLibraryType": null,
>     "stockFlag": false,
>     "stockSort": null,
>     "firstSortName": null,
>     "secondSortName": null,
>     "componentBrandList": [],
>     "searchSource": "search",
>     "componentSpecificationList": [],
>     "componentAttributeList": [],
>     "paramList": [],
>     "startStockNumber": null,
>     "endStockNumber": null
> }
> 
> This gives a resulting JSON like this
> 
> {
>     "code": 200,
>     "data": {
>         "componentPageInfo": {
>             "total": 1,
>             "list": [
>                 {
>                     "componentCode": "C1034",
>                     "componentBrandEn": "Sunlord",
>                     "componentModelEn": "SDFL1608Q4R7KTF",
>                     "componentImageUrl": "",
>                     "productBigImageAccessId": "8552460562370867200",
>                     "dataManualUrl": "https://www.lcsc.com/datasheet/lcsc_datasheet_2310301641_Sunlord-SDFL1608Q4R7KTF_C1034.pdf",
>                     "dataManualFileAccessId": "8579705830068764672",
>                     "dataManualOfficialLink": null,
>                     "componentName": "Sunlord SDFL1608Q4R7KTF",
>                     "minImage": "",
>                     "minImageAccessId": "8552460562958340097",
>                     "imageList": [
>                         {
>                             "productImage": null,
>                             "productImageAccessId": null,
>                             "productBigImage": null,
>                             "productBigImageAccessId": null,
>                             "fileType": null
>                         },
>                         {
>                             "productImage": null,
>                             "productImageAccessId": null,
>                             "productBigImage": null,
>                             "productBigImageAccessId": null,
>                             "fileType": null
>                         },
>                         {
>                             "productImage": null,
>                             "productImageAccessId": null,
>                             "productBigImage": null,
>                             "productBigImageAccessId": null,
>                             "fileType": null
>                         },
>                         {
>                             "productImage": null,
>                             "productImageAccessId": null,
>                             "productBigImage": null,
[... 123 lines cut ...]
>             "isFirstPage": false,
>             "isLastPage": false,
>             "hasPreviousPage": false,
>             "hasNextPage": false,
>             "navigatePages": 0,
>             "navigatepageNums": null,
>             "navigateFirstPage": 0,
>             "navigateLastPage": 0
>         },
>         "sortAndCountVoList": [],
>         "queryString": "SDFL1608Q4R7KTF"
>     },
>     "message": null
> }
> 
> This data includes `"preferredComponentFlag": true` which I belive is what we are looking for.
> 
> But, this API endpoint is protected by some sort of cloudfront as I get errors when trying to replicate it via curl
> 
> {"code":500,"data":null,"message":"System error. Please try again later."}       
> 
> So I have no idea how to get this flag the easy way and without sending a shitload of requests to JLC which might upsets them

Thank You for looking into it.
Maybe its possible to get more info from: https://jlcpcb.com/parts/basic_parts?

neggles, 2025-07-02:
@Bouni from some tests I've just run it seems like it works fine without auth? you just can't reuse a client object for subsequent requests, for some reason. But you can dump the whole thing in one go;

```py
from pathlib import Path
from httpx import Headers
from httpx import request
import json

work_dir = Path(__file__).parent.resolve() if "__file__" in locals() else Path.cwd()

api_url = "https://jlcpcb.com/api/overseas-pcb-order/v1/shoppingCart/smtGood/selectSmtComponentList/v2"
req_headers = Headers(
    {
        "authority": "jlcpcb.com",
        "accept": "application/json, text/plain, */*",
        "accept-encoding": "gzip, deflate",
        "accept-language": "en-US,en;q=0.9,zh-CN;q=0.7,zh;q=0.6",
        "cache-control": "no-cache",
        "content-type": "application/json",
        "pragma": "no-cache",
        "referer": "https://jlcpcb.com/parts/basic_parts",
        "sec-ch-ua": '"Microsoft Edge";v="137", "Chromium";v="137", "Not A Brand";v="24"',
        "sec-ch-ua-platform": '"Windows"',
    }
)
req_payload = {
    "componentAttributes": [],
    "componentBrand": None,
    "componentLibraryType": "",
    "componentLibraryTypeCheck": False,
    "componentSpecification": None,
    "currentPage": 1,
    "pageSize": 10000,
    "preferredComponentFlag": True,
    "preferredComponentFlagCheck": True,
    "searchSource": "search",
    "stockFlag": None,
    "stockSort": None,
}

resp = request("POST", api_url, headers=req_headers, json=req_payload)
resp.raise_for_status()

data = resp.json()
# dump the response to a file for debugging
print("Saving raw response to response.json")
Path.cwd().joinpath("response.json").write_text(
    json.dumps(data, indent=4, ensure_ascii=False, default=str), encoding="utf-8"
)

part_list = data.get("data", {}).get("componentPageInfo", {}).get("list", [])
if part_list:
    print(f"Found {len(part_list)} components, sorting by componentId.")
    # Sort the part list by componentId
    part_list.sort(key=lambda x: x.get("componentId", 0))

print("Saving part list to part_list.json")
Path.cwd().joinpath("part_list.json").write_text(
    json.dumps(part_list, indent=4, ensure_ascii=False, default=str), encoding="utf-8"
)

print("Done.")
```

seems to work fine.