Bouni/kicad-jlcpcb-tools #557: No components in the table

valerionew, 2024-12-06:
Hey, I'm a new user of the plugin, just installed it! Maybe there is some crucial information that I am not finding, but I have been looking around for a while and couldn't find anything. Apologies in advance if this is trivial!

**Describe the bug**
When the plugin is opened, no components are there. I have installed the plugin from the manager.

This is my log. Maybe it is relevant, maybe not, but it seems that the order of things in the log is a bit off.
```
2024.12.06 14:26:35 - DEBUG - download -  Parts db is split into 15 parts. Proceeding to download...
2024.12.06 14:26:35 - DEBUG - download -  Removing any spurious old zip part files...
2024.12.06 14:26:36 - DEBUG - download -  Parts db is split into 15 parts. Proceeding to download...
2024.12.06 14:26:36 - DEBUG - download -  Removing any spurious old zip 2024.12.06 14:28:55 - DEBUG - download -  Parts db is split into 15 parts. Proceeding to download...
2024.12.06 14:28:55 - DEBUG - download -  Removing any spurious old zip part files...
2024.12.06 14:28:55 - DEBUG - download -  Removing /home/vnappi/.local/share/kicad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/jlcpcb/parts-fts5.db.zip.001.
2024.12.06 14:29:07 - DEBUG - download -  Parts db is split into 15 parts. Proceeding to download...
2024.12.06 14:29:07 - DEBUG - download -  Removing any spurious old zip part files...
2024.12.06 14:29:07 - DEBUG - download -  Removing /home/vnappi/.local/share/kicad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/jlcpcb/parts-fts5.db.zip.001.
part files...
2024.12.06 14:26:36 - DEBUG - download -  Removing /home/vnappi/.local/share/kicad/8.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/jlcpcb/parts-fts5.db.zip.001.
2024.12.06 14:25:37 - INFO - setup -  Data directory 'jlcpcb' does not exist and will be created.
2024.12.06 14:25:37 - DEBUG - create_rotation_table -  Create SQLite table for rotations

```
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Open the plugin
2. Notice that no components are there

**Expected behavior**
Components are expected to be in the table.

# KiCad Version

```
Application: KiCad PCB Editor x86_64 on x86_64

Version: 8.0.6, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.0.1
	FontConfig 2.15.0
	libcurl/8.11.0 OpenSSL/3.4.0 zlib/1.3.1 brotli/1.1.0 zstd/1.5.6 libidn2/2.3.7 libpsl/0.21.5 libssh2/1.11.0 nghttp2/1.64.0 nghttp3/1.6.0

Platform: Arch Linux, 64 bit, Little endian, wxGTK, X11, gnome-xorg, x11
OpenGL: Intel, Mesa Intel(R) Xe Graphics (TGL GT2), 4.6 (Compatibility Profile) Mesa 24.2.7-arch1.1

Build Info:
	Date: Oct 16 2024 20:40:29
	wxWidgets: 3.2.6 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.86.0
	OCC: 7.8.1
	Curl: 8.10.1
	ngspice: 43
	Compiler: GCC 14.2.1 with C++ ABI 1019

Build settings:
	KICAD_USE_EGL=ON

```

Bouni, 2024-12-16:
I think you have to repopen the plugin.
For some reson if you open it the first time it download the database and does not read the parts from the board.
On the second opening, it does that and you're good to go.

Let me know if that works.

Bouni, 2024-12-16:
Duplicate of #550

Bouni, 2024-12-16:
I changed the download and unzip code just days ago, now you get visual feedback in the prgress bar at the bottom while the download and unzip is ongoing (log messages as well).

valerionew, 2024-12-16:
I think I tried the version with the bar already, but the bar was super jumpy and flickery, so maybe that's why I may have overlooked it.

Would it be possible to move the bar inside the download button instead? To me, that would be a more straight-forward indication that the download is ongoing.

nickoe, 2025-03-19:
I noticed the same today on a windows build of kicad 9.0.0 and plugin 2025.03.01.  Closing and reopening resolved the empty table issue :S

z2amiller, 2026-04-10:
> I think I tried the version with the bar already, but the bar was super jumpy and flickery

The 'flickery download bar' and the log out of order is almost certainly
Fixed by: #719

(The self.store issue mentioned by nickoe@ isn't the same issue and is a duplicate of #507,
although there are duplicate log messages in there that suggest that this actually could be a
race condition that would also be fixed by that PR)

closed 2026-04-10 by z2amiller