Bouni/kicad-jlcpcb-tools #561: TypeError: 'SwigPyObject' object is not iterable on KiCad  kicad-unified-universal-20250101-1013-6c43c25b4c

NikolayMurha, 2025-01-02:
**Describe the bug**
When I try to open plugin I have error:

```
Traceback (most recent call last):

File "/Users/nikolay/Documents/KiCad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 28, in Run
dialog = JLCPCBTools(None)

File "/Users/nikolay/Documents/KiCad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 497, in __init__
self.init_store()

File "/Users/nikolay/Documents/KiCad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 521, in init_store
self.store = Store(self, self.project_path, self.pcbnew.GetBoard())

File "/Users/nikolay/Documents/KiCad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/store.py", line 34, in __init__
self.update_from_board()

File "/Users/nikolay/Documents/KiCad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/store.py", line 178, in update_from_board
"lcsc": get_lcsc_value(fp),

File "/Users/nikolay/Documents/KiCad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/helpers.py", line 121, in get_lcsc_value
for field in fp.GetFields():

TypeError: 'SwigPyObject' object is not iterable
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install the latest nightly build ( kicad-unified-universal-20250101-1013-6c43c25b4c)
2. Install plugin   'Bouni/kicad-jlcpcb-tools'
4. Open plugin and see error

# KiCad Version

```
Application: KiCad PCB Editor arm64 on arm64

Version: 9.0.0-rc1-273-g6c43c25b4c, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.1.0
	FontConfig 2.15.0
	libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.63.0

Platform: macOS Sequoia Version 15.2 (Build 24C101), 64 bit, Little endian, wxMac
OpenGL: Apple, Apple M1 Pro, 2.1 Metal - 89.3

Build Info:
	Date: Jan  1 2025 08:09:04
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.87.0
	OCC: 7.8.1
	Curl: 8.7.1
	ngspice: 44
	Compiler: Clang 16.0.0 with C++ ABI 1002
	KICAD_IPC_API=ON

```

markus-k, 2025-01-07:
Try a more recent nightly, this was a bug and has been fixed in KiCad.

chmorgan, 2025-03-29:
Closing due to inactivity.

closed 2025-03-29 by chmorgan