Bouni/kicad-jlcpcb-tools #563: JLCPCB Fabrication error - unsupported file extensions

AC8L, 2025-01-19:
**Describe the bug**
This plugin is my default for generating fabrication files for the JLCPCB orders. On my recent order two holes for the footprint were not fabricated although JLPCB's Gerber viewer showed them.

I opened a case and support engineer told me that according to their investigation their fabrication software was confused by file extensions in my gerber zip file. Engineer said that instead of all files having .gbr file extensions files must be names according to the following scheme:
https://jlcpcb.com/help/article/Suggested-Naming-Patterns

VScore file and Paste files MUST NOT be included.

And, JLCPCB refused to reimburse the cost due to unsupported fabrication file format submission.

Without modifying the design I manually changed file extensions and submitted to JLCPCB again. All holes were properly cut this time.

Can you add to the script automatic file extension naming please? Make sure 4-layer and 6-layer extensions are handled properly too.

**To Reproduce**
If needed I can submit gerber files for the erroneous order.

**Expected behavior**
All holes in the PCB have to be cut as shown in gerber viewer either in KiCad or JLCPC site's gerber viewer.

# KiCad Version
Application: KiCad arm64 on arm64

Version: 8.0.8, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.1.0
	FontConfig 2.15.0
	libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.63.0

Platform: macOS Sequoia Version 15.2 (Build 24C101), 64 bit, Little endian, wxMac

Build Info:
	Date: Jan 12 2025 21:41:46
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.87.0
	OCC: 7.8.1
	Curl: 8.7.1
	ngspice: 44
	Compiler: Clang 16.0.0 with C++ ABI 1002

Build settings:

```
```

Bouni, 2025-01-20:
I'm sorry that your first order turned out to be scrap beacuse of this.
JLC gets more and more picky over time, their customer service degrades compared to some years ago ...

I'll see if I can easily fix this.

closed 2025-01-20 by Bouni

AC8L, 2025-01-20:
[image]

Bouni,

Thank you very much for removing score and paste files. I am attaching the communication with JLCPCB customer service where they directly refer to the file extension as a root cause of not drilling the hole. And they provided a link to naming extension that I shared. I understand that both of us got different replies from different agents which of course speaks of JLCPCP degrading customer service. However in my case just following there suggestion and renaming files got order fabricated right second time. It is unfortunate that you will not script this. Because to be assured of proper fabrication I will have to manually rename fabrication files every time they are generated. I hope others are not going to have the same issue.

Bouni, 2025-02-03:
@AC8L In my opinion this was an excuse from JLC to cover up a mistake on their side. I ordered dozens of pcbs so far and never had an issue.
You can also check the "verify productin file" checkbox to make absolutely sure you get what you expect:

[image]

I know that adds 48 hours extra, but if you don't trust them its maybe worth it.

AC8L, 2025-02-03:
I have had the same experience as you - ordered many PCBs. And this was the first incident. I always check "verify production file". In both cases - with erroneous order and correct order the verification picture showed holes to be fabricated.

I understand they might use this as an excuse. And even ready that for their every fabrication error they are going to find an excuse in order not to reimburse the customer.

The only fact that gives me a caution is that just by renaming file extensions according to customer service recommendation I got proper fabrication. Of course, they could monitor my specific orders as of a "person of interest". Even though they have  thousands of orders submitted etc. But that is what doctors call paranoia :)

Anyway, I keep manually renaming gerber files after the are produced by your plugin before submission to the JLCPCB. I hope there will be no errors anymore. Or let's see what kind of excuse they will give next time.