Bouni/kicad-jlcpcb-tools #568: some components just have * as a value

steeley, 2025-01-25:
Sometimes components have * as their value. Even if I put the correct value in the "value" field, JLCPCB tools does not update.
On large projects I have to manually edit the CSV file otherwise it is not accepted for manufacture.

This is now a major problem as it makes everything very slow.

I'm on Kicad 8.08 OSX intel

Bouni, 2025-02-03:
Can you give an example of what components you mean?

steeley, 2025-02-03:
It's always resistors and capacitors from JLC library. Here are some that don't show values correctly, there are more!:
part no:
C23630
C22978
C82282
C1671
C22978
C22961
C25744
C2582

steeley, 2025-02-26:
The issue is with components such as capacitors  that have a * as the value but have a field that says 'capacitance' (10µF for example) Even when you go to the schematic and fill in the value, the value does not update in the tools plugin.
So the plugin is of no use - the BOM has to be done manually.

[image]

chmorgan, 2026-09-06:
@steeley can you clarify your use case? You have an empty kicad 'value' field but an additional 'capacitance' field that has a value?

z2amiller, 2026-09-26:
I don't think we can support this - it sounds like you've used the 'Value' field to mean something else, and put the actual value in a different field in your schematic.  I spot checked the parts you mentioned and they're in the index and have searchable values.

closed 2026-09-26 by z2amiller