Bouni/kicad-jlcpcb-tools #570: Some components not loading pictures anymore - includes root cause

markusdd, 2025-02-04:
Hi,

I noticed in my own tool, that for some components (e.g. C55136) pictures are not loading anymore.
The root cause is that for some components the picture URLs are now empty:

endic-sam928281, 2025-02-04:
Hello, we tried to solve the issue.

This is what we did:

Modified the `get_part_data` method in the `PartDetailsDialog` class to handle cases where the picture URL is empty. Added a fallback to use a placeholder image when no picture URL is available.

You can review changes in this commit: https://github.com/endic-sam928281/Bouni-kicad-jlcpcb-tools-570/commit/c3161196ab6a264d5c7e49501807426c93529e03.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

markusdd, 2025-02-04:
Yes, but there is no need to use placeholders. The pictures are available, you just need to use a different route.

PR #576 (2025-03-05): Fix part picture download
This fixes #570 as suggested by @markusdd
closed 2025-03-05 by Bouni