Bouni/kicad-jlcpcb-tools #571: part number fields seem to be ignored

steeley, 2025-02-08:
if I add a part and put a custom field as in the picture below, the part numbers get ignored and the plug-in substitutes  a part from the database.
This used to work I'm sure. I do my boards this way and it is now a big problem. ! The plug-in is now un usable for me along with other issues . Sorry!

[image]

Application: KiCad Schematic Editor x86_64 on x86_64

Version: 8.0.8, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.1.0
	FontConfig 2.15.0
	libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.51.0

Platform: macOS Ventura Version 13.7.3 (Build 22H417), 64 bit, Little endian, wxMac
OpenGL: ATI Technologies Inc., AMD Radeon Pro 560X OpenGL Engine, 2.1 ATI-4.14.1

Build Info:
	Date: Jan 12 2025 22:32:39
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.87.0
	OCC: 7.8.1
	Curl: 8.7.1
	ngspice: 44
	Compiler: Clang 16.0.0 with C++ ABI 1002

Build settings:

Bouni, 2025-03-05:
I can't reproduce your problem here ...

1 . I add a custom LCSC field in the schematic

[image]

2. I update the pcb from the schematic and verify the value is in the footprint proprties

[image]

3. I open the plugin

[image]

Is that what you did?

steeley, 2025-03-05:
have tried again. I updated my my Kicad. It seems ok now. I think the problem might have been I had 2 plugin windows open and this stopped it working. thanks!

closed 2025-03-25 by Bouni