Bouni/kicad-jlcpcb-tools #572: New to KiCAD, but cannot see this tool

jds11111, 2025-02-11:
---------Add your issue details below----------- -->

**Describe the bug**
Installed KiCAD from standard repo, v6.0, and then from updated rep, v8.0 on Ubuntu. When I add this 3rd party tool, either through the GUI, or via git, it never shows up in the PCB editor. I should warn you that I am just learning.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Tools->Plugin Manager
2. Click on Manager Repositories, add this one, install, apply, etc. Claims that everything is done.
3. Open PCB editor, see the PCB image
4. Go to Tools, and scroll down to External Plugins, but ain't nothin' there. 

**Expected behavior**
That is with the GUI, but using git in the plugins folder doesn't work either. Closing the software and restarting makes no difference

# KiCad Version

```
Application: KiCad x86_64 on x86_64

Version: 8.0.8-8.0.8-0~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.18

Platform: Ubuntu 22.04.5 LTS, 64 bit, Little endian, wxGTK, X11, KDE, x11
OpenGL: Intel, Mesa Intel(R) HD Graphics 2000 (SNB GT1), 3.3 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3

Build Info:
	Date: Jan 14 2025 02:26:25
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.81.0
	ngspice: 43
	Compiler: GCC 11.4.0 with C++ ABI 1016

Build settings:

```

chmorgan, 2025-03-29:
Hmm. No idea why it might not be working for you. Any errors reported when installing the plugin?

jds11111, 2025-04-25:
So, I just tried installing again, but the install button stays greyed out with Install Pending.

Ragarnoy, 2025-07-28:
Hey, @Bouni I just installed and had the exact same issue, should I open a new one? I have installed KiCAD and the plugin "normally", and I haven't seen a single error anywhere