Bouni/kicad-jlcpcb-tools #573: Support for Kicad v9

filipeaguiar22, 2025-02-26:
**Is your feature request related to a problem? Please describe.**
I would like to know if you foresee the support for the newly release Kicad v9

**Describe the solution you'd like**
This is a great tool, that simplifies PCB/PCB assembly ordering, from JCPCB. 

**Describe alternatives you've considered**
At the moment, the plugin does not work with the new version.

andy9a9, 2025-02-27:
Is there some kind of error, or what's really not working?

filipeaguiar22, 2025-02-27:
Hi Andy,

Thank you for the quick reply!
Indeed there is one. 
I submitted a 4 layer board to JLCPB and they pointed out that GERBER output for the Internal layers show up as solder mask and ithe geometry differs from the actual In1.Cu or In2.Cu.
I attached the nice error detection performed by JLCPCB.
As of now, I still use the plugin, for BOM and Placcement, but I use the Kicad Plot and Drill to generate the gerbers.

[image]

[image]

andy9a9, 2025-02-27:
Hello
Correct me if I am wrong, but as I can read the answer from JLC, they are complaining, that you ordered the 2-layers board and uploaded the 4-layers, or?

Nevertheless maybe it's only a part of the communication. So the selected file on the left side is corresponding with the picture on the right side, right? Maybe a other question, if you generate the this Gerber file directly from kicad is it ok, right?

netkruzer, 2025-02-28:
I'm having the same issue. I export my 4 layer board with the JLCPCB plugin from KiCAD 9.0, upload, and JLC's website sees it as a 2 layer board.

I export using KiCAD's plot and drill and JLC's website sees it as a 4 layer board.

I'm glad I caught this, I need those inside layers!

Alectardy98, 2025-03-01:
In the current kicad nightly, I can not generate files from the plugin in mac os or arch linux. I am not sure if there are plans to support the current nightly's but it would be great to use this plugin again

wiserst, 2025-03-01:
Today I found out, when I wanted to submit production to JLCPCB, that in the stable version of KiCad 9.0.0, even the generation of BOM and CPL files no longer works. The directories are empty. Please, can you update such a great plugin so that it works again? I can't do without it anymore. Thank you.

andy9a9, 2025-03-01:
Let's wait for review my PR. Hopefully it's fixed there ;).

filipeaguiar22, 2025-03-01:
You were fast!
Many thanks!

wiserst, 2025-03-02:
Thank you. I tried it and the plugin generates files again. I tried importing GERBER, BOM and CPL into JLCPCB. Everything seems to be OK.

PR #575 (2025-03-05): Fix/layers
- fix typo
  - `Silk top` -> `Silk bottom`
- replace generated layer numbers with pcbnew's enums
  - The layer IDs have been changed in KiCad 9, so get the enums directly from `pcbnew`. This change is backport compatible with the older KiCad versions.

Closes #573, closes #567
closed 2025-03-05 by Bouni