Bouni/kicad-jlcpcb-tools #574: I'm considering stepping down as maintainer.

whmountains, 2025-02-26:
Hi @Bouni 

Several months ago I was contributing regularly and you trusted me with commit rights.  Unfortunately the time I had to contribute regularly has dried up.

Recently my company introduced a policy to allow employee-chosen open source sponsorships.  The amounts are not large, maybe $20-30 per month, but I would like to sponsor this project since it has become important part of my workflow.

Sadly one of the requirements to sponsor is that I not be personally involved in the project.  Therefore I would need to step down as a maintainer before commencing sponsorship.

Let me know whether you'd like me to stay as a maintainer, and potentially contribute in the future, or officially step down and send some money your way instead.

Bouni, 2025-02-28:
Hi @whmountains ,

After thinking about this for a moment, I think I would like it better if you spend the budget your company provides on some other projects and instead coninue to support this project with contributions.

As everyone has noticed, my spare time is extremely tight and therefore the development has kind of stalled, which does not make me happy to be honest.

I concider contributions and maintainig the project so much more valuable than money and hope yo continue to support this a sa maintainer 😊

whmountains, 2025-03-11:
Thank you for the confidence!  I hope to be able to contribute.

closed 2025-03-11 by whmountains