Bouni/kicad-jlcpcb-tools #578: "0Ω" returns 0 results

nippoo, 2025-03-10:
**Describe the bug**
Searching for "0Ω" or "0Ω 0805" or "0Ω 0402" returns no results, even though "10Ω 0402" (for example) returns correctly

**To Reproduce**
Steps to reproduce the behavior:
Select any part (or a resistor with value "0") press "Assign LCSC number", make sure the search bar begins with "0Ω"

**Expected behavior**
Parts should return e.g. C17477

# KiCad Version
```
Application: KiCad arm64 on arm64
Version: 9.0.0, release build
Libraries:
    wxWidgets 3.2.6
    FreeType 2.13.3
    HarfBuzz 10.2.0
    FontConfig 2.15.0
    libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.62.0
Platform: macOS Sequoia Version 15.1.1 (Build 24B91), 64 bit, Little endian, wxMac
OpenGL: Apple, Apple M1 Pro, 2.1 Metal - 89.3
Build Info:
    Date: Feb 19 2025 13:05:44
    wxWidgets: 3.2.6 (wchar_t,wx containers)
    Boost: 1.87.0
    OCC: 7.8.1
    Curl: 8.7.1
    ngspice: 44.2
    Compiler: Clang 16.0.0 with C++ ABI 1002
    KICAD_IPC_API=ON
```

chmorgan, 2025-03-25:
I thought I knew what the issue is here but I’m still trying to debug it. Same query works locally on the same database.

I’ll msg here when I have a PR that fixes this.

chmorgan, 2025-03-26:
@nippoo alright, this is a bug in the version of sqlite used by kicad. I'm not sure exactly when it was fixed but the same query that produces no results with the plugin produces results when run locally with a newer/different version of sqlite.

I'll continue to dig into things a bit further to find the root cause.

Bouni, 2025-06-10:
It seems that this works as expected (despite not like I wish it would work) on Windows using 9.0.2

[image]

silicongarage, 2025-07-22:
I have the same problem.   Any single digit search (0~9) fails to show any values.

OS:  MacOS
kicad: 9.0.3
jlcpcb-tools:  2025.04.02  (installed from plugin manager)

silicongarage, 2025-07-22:
It's interesting because  two (or more) digits with the ohms symbol work correctly

nippoo, 2025-07-22:
Hey @Bouni - sorry for the delay! Can confirm this still isn't fixed.

Debug output for 0Ω and 10Ω:

```
2025.07.22 12:08:03 - DEBUG - search -  query 'SELECT "LCSC Part","MFR.Part","Package","Solder Joint","Library Type","Stock","Manufacturer","Description","Price","First Category" FROM parts WHERE parts MATCH '"0402"' AND description LIKE '%0Ω%' AND "Library Type" IN ("Basic","Extended") ORDER BY "LCSC Part" COLLATE naturalsort ASC LIMIT 1000'
2025.07.22 12:08:03 - DEBUG - search -  len(result) 0
2025.07.22 12:08:10 - DEBUG - search -  query 'SELECT "LCSC Part","MFR.Part","Package","Solder Joint","Library Type","Stock","Manufacturer","Description","Price","First Category" FROM parts WHERE parts MATCH '"10Ω" AND "0402"' AND "Library Type" IN ("Basic","Extended") ORDER BY "LCSC Part" COLLATE naturalsort ASC LIMIT 1000'
2025.07.22 12:08:10 - DEBUG - search -  len(result) 3
```

sqlite version etc if it's helpful or different on Mac builds:
```
2025.07.22 12:07:50 - DEBUG - __init__ -  partsdb_file /Users/nippoo/Documents/KiCad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/jlcpcb/parts-fts5.db
2025.07.22 12:07:50 - DEBUG - __init__ -  sqlite.sqlite_version 3.37.2
2025.07.22 12:07:50 - INFO - setup -  Data directory '/Users/nippoo/Documents/KiCad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/jlcpcb' exists, not creating
2025.07.22 12:07:50 - DEBUG - init_library -  JLCPCB version 2025.04.02, last database update 2025-07-22T05:21:46.337241, part count 6949739, size (bytes) 4680962048
```

nippoo, 2025-07-22:
It's also worth noting that the 10Ω 0402 result only returns 3 items, whereas there are hundreds on the JLCPCB website.

"0 0402" returns 34 results, all unhelpful
"10 0402" returns 11 unhelpful results
"0Ω 0402" returns no results
"10Ω 0402" returns three results, all of which are extended components with little stock, and fails to match C25077 (the actual basic 10Ω 0402 resistor).
"0Ω" returns 0 results
"10Ω" returns 220 results, but no basic parts and mostly out of stock.

It feels like this has actually been a regression... as I can't actually find basic parts for common values any more with or without the ohm symbol.

nippoo, 2025-07-22:
So I think there's a secondary issue here, I'll file another bug.

The LIKE vs MATCH logic may also still not be working properly but it's hard to tell right now...

z2amiller, 2026-09-20:
fixed by #728 

(plus some of the other reports were the upstream database corruption)

closed 2026-09-20 by z2amiller