Bouni/kicad-jlcpcb-tools #579: "Export to schematic" doesn't work in KiCad 9

nippoo, 2025-03-10:
**Describe the bug**
Exporting to schematic appears to work, but fails to update the LCSC field

**To Reproduce**
Press "Export to Schematic" after editing values

**Expected behavior**
Values should update

# KiCad Version
```
Application: KiCad arm64 on arm64
Version: 9.0.0, release build
Libraries:
    wxWidgets 3.2.6
    FreeType 2.13.3
    HarfBuzz 10.2.0
    FontConfig 2.15.0
    libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.62.0
Platform: macOS Sequoia Version 15.1.1 (Build 24B91), 64 bit, Little endian, wxMac
OpenGL: Apple, Apple M1 Pro, 2.1 Metal - 89.3
Build Info:
    Date: Feb 19 2025 13:05:44
    wxWidgets: 3.2.6 (wchar_t,wx containers)
    Boost: 1.87.0
    OCC: 7.8.1
    Curl: 8.7.1
    ngspice: 44.2
    Compiler: Clang 16.0.0 with C++ ABI 1002
    KICAD_IPC_API=ON
```

Bouni, 2025-03-25:
@computergeek1507 Any chance you can take a look at this?

andy9a9, 2025-03-27:
Isn't #580 fixing this?

chmorgan, 2025-03-29:
@nippoo can you see if the latest 'main' of the plugin resolves the issue for you? I merged the changes from @andy9a9's PR yesterday and testing here indicates the v8+ schematic update code is  being called (although I don't use the schematic export feature so I'm not sure what its supposed to do).

chmorgan, 2025-04-02:
@nippoo 2025.04.02 has been released, please try this and re-open if your issue hasn't been resolved.

closed 2025-04-02 by chmorgan

andy9a9, 2025-07-09:
> Still not working for me

Do yo have the latest version, right? Some kind logs or more info about OS would be welcome :)

slabua, 2025-07-09:
@andy9a9 I suppose I have as there is no update notification in KiCad main page.

This is the About from KiCad

```
Application: KiCad x86_64 on x86_64

Version: 9.0.2-2.fc42, release build

Libraries:
	wxWidgets 3.2.8
	FreeType 2.13.3
	HarfBuzz 10.4.0
	FontConfig 2.16.0
	libcurl/8.11.1 OpenSSL/3.2.4 zlib/1.3.1.zlib-ng brotli/1.1.0 libidn2/2.3.8 libpsl/0.21.5 libssh/0.11.2/openssl/zlib nghttp2/1.64.0 OpenLDAP/2.6.9

Platform: Fedora Linux 42 (Workstation Edition), 64 bit, Little endian, wxGTK, X11, gnome, wayland
OpenGL: Intel, Mesa Intel(R) HD Graphics 4400 (HSW GT2), 4.6 (Compatibility Profile) Mesa 25.1.4

Build Info:
	Date: May 16 2025 00:00:00
	wxWidgets: 3.2.8 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.83.0
	OCC: 7.8.1
	Curl: 8.11.1
	ngspice: 44.2
	Compiler: GCC 15.1.1 with C++ ABI 1020
	KICAD_IPC_API=ON

Locale: 
	Lang: en_GB
	Enc: UTF-8
	Num: 1,234.5
	Encoded кΩ丈: D0BACEA9E4B888 (sys), D0BACEA9E4B888 (utf8)
```

andy9a9, 2025-07-09:
> I just noticed something, in the export log it says Kicad 7 instead of 9. Is it making a mistake recognising the version or?
> 
> ```
> 2025.07.09 22:36:42 - INFO - load_schematic -  Kicad 7...
> 2025.07.09 22:36:42 - INFO - _update_schematic7 -  Reading /home/xxx/xxx.kicad_sch...
> 2025.07.09 22:36:42 - INFO - _update_schematic7 -  Added LCSC's to /home/xxx/xxx.kicad_sch (maybe?)
> ```
Yes, that's wrong, while it has to be `Kicad 8+...`. So maybe you hit some error from https://github.com/Bouni/kicad-jlcpcb-tools/pull/611 .  So would it be possible to test it again on the latest changes from `master`? Other option is to add some prints to see what `GetBuildVersion()` is really returning.

slabua, 2025-07-10:
@andy9a9 to test something, i did a dirty trick and it worked. I commented the version check in `schematicexport.py`
Indeed GetBuildVersion() causes some issues, at least with my fedora build of kicad

```
    def load_schematic(self, paths):
        """Load schematic file."""
        """
        if is_version6(GetBuildVersion()):
            self.logger.info("Kicad 6...")
            for path in paths:
                self._update_schematic6(path)
        elif is_version7(GetBuildVersion()):
            self.logger.info("Kicad 7...")
            for path in paths:
                self._update_schematic7(path)
        else:
        """
        self.logger.info("Kicad 8+...")
        for path in paths:
            self._update_schematic(path)
```

slabua, 2025-07-18:
@andy9a9 thanks it's working