Bouni/kicad-jlcpcb-tools #582: "Generate" doesn't work in KiCad 9

nickoe, 2025-03-21:
**Describe the bug**
After I made the changes described in 
https://github.com/Bouni/kicad-jlcpcb-tools/pull/580#issuecomment-2743054825

To be able to load the plugin and assign LCSC numbers. I now tried to hit "Generate" but nothing happens. 

In terminal I see this exception:
```
2025.03.21 12:26:52 - DEBUG - update_from_board -  Part R1 does not exist in the database and will be created from the board.
Traceback (most recent call last):
  File "C:\Users\foo\Documents\KiCad\9.0\scripting\plugins\kicad-jlcpcb-tools\mainwindow.py", line 870, in generate_fabrication_data
    self.fabrication.generate_geber(layer_count)
  File "C:\Users\foo\Documents\KiCad\9.0\scripting\plugins\kicad-jlcpcb-tools\fabrication.py", line 119, in generate_geber
    pctl = PLOT_CONTROLLER(self.board)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\KiCad\9.0\bin\Lib\site-packages\pcbnew.py", line 8084, in __init__
    _pcbnew.PLOT_CONTROLLER_swiginit(self, _pcbnew.new_PLOT_CONTROLLER(aBoard))
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: in method 'new_PLOT_CONTROLLER', argument 1 of type 'BOARD *'
```

**To Reproduce**
Steps to reproduce the behavior:
1. Open JLCPCB Tools in a design with a part
2. Click on "Generate"
3. Note no fabrication output is generated
4. If you start it from a shell you see the above backtrace.

**Expected behavior**
No exception and production files being generated.
# KiCad Version

```
```

nickoe, 2025-03-21:
@andy9a9  Do you see this issue as well?

nickoe, 2025-03-21:
In another PCB I get:
```
Traceback (most recent call last):
  File "C:\Users/noestergaard/Documents/KiCad/9.0/scripting\plugins\kicad-jlcpcb-tools\mainwindow.py", line 870, in generate_fabrication_data
    self.fabrication.generate_geber(layer_count)
  File "C:\Users/noestergaard/Documents/KiCad/9.0/scripting\plugins\kicad-jlcpcb-tools\fabrication.py", line 136, in generate_geber
    popt.SetPlotInvisibleText(False)
    ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetPlotInvisibleText'
```

And here it looks like that method is gone, was available in:
https://docs.kicad.org/doxygen-python-8.0/classpcbnew_1_1PCB__PLOT__PARAMS.html

Gone in:
https://docs.kicad.org/doxygen-python-9.0/classpcbnew_1_1PCB__PLOT__PARAMS.html

andy9a9, 2025-03-21:
> [@andy9a9](https://github.com/andy9a9) Do you see this issue as well?

Nope. I clicked on Generate button and the gerbers have been generated fine.

Bouni, 2025-03-27:
I have ordered several boards out of KiCAD 9 as well, no issues so far ...

```
Application: KiCad x64 on x64

Version: 9.0.0, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.2.0
	FontConfig 2.15.0
	libcurl/8.11.1-DEV Schannel zlib/1.3.1

Platform: Windows 11 (Erzeugungsversion 26100), 64-bit Edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb 19 2025 17:46:53
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.86.0
	OCC: 7.8.1
	Curl: 8.11.1-DEV
	ngspice: 44
	Compiler: Visual C++ 1942 without C++ ABI
	KICAD_IPC_API=ON

Locale: 
	Lang: de_CH
	Enc: UTF-8
	Num: 1’234.5
	Encoded кΩ丈: D0BACEA9E4B888 (sys), D0BACEA9E4B888 (utf8)
```
This is the version info of my KiCAD installation.

I see you have a 9.0.1 RC1 installed, seems they start dropping stuff.

I'll see if I can add a workaround

chmorgan, 2025-03-27:
@Bouni I sent you an email yesterday with a proposed change idea. If you haven't received it can you email me with your email address and I'll respond? Discussions isn't enabled here and it seemed like a good way to chat vs. opening an issue etc.

I've also been looking into this a little bit as I have a design waiting to be jlcpcb that is stuck behind being able to generate the design package.

chmorgan, 2025-03-28:
@nickoe I'm going to take a look at this over the next few days. I'm held up by this not working smoothly with 9.x versions.

chmorgan, 2025-03-28:
@nickoe and others, please try the changes in #590, that lets me generated here with v9.0.1-rc1

chmorgan, 2025-03-30:
Merged #590, please reopen if this doesn't resolve the issue for you.

closed 2025-03-30 by chmorgan