Bouni/kicad-jlcpcb-tools #585: Sorting columns are incorrectly handled in the part selector

gyohng, 2025-03-26:
**Describe the bug**

[image]
After 'Params' header, the sorting behaviour is off-by-one. Clicking Params header sorts by Stock, clicking Stock header sorts by Manufacturer, etc.

**To Reproduce**
Steps to reproduce the behavior:
1. Click Assign LCSC number
2. Type something to display search results
3. Try sorting results by clicking different table headers (Params, Stock, Manufacturer, Description)

**Expected behavior**
Clicked columns should sort themselves and not the other columns.

# KiCad Version

KiCad version is irrelevant. I think it'll be okay to add a comment in the source code, which says something like _'# If you add more columns, don't forget to update the column order list found in function X'_

```
Application: KiCad arm64 on arm64

Version: 9.0.1-rc2, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.2.0
	FontConfig 2.15.0
	libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.63.0

Platform: macOS Sequoia Version 15.3.2 (Build 24D81), 64 bit, Little endian, wxMac
OpenGL: Apple, Apple M1 Max, 2.1 Metal - 89.3

Build Info:
	Date: Mar 26 2025 00:39:58
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.87.0
	OCC: 7.8.1
	Curl: 8.7.1
	ngspice: 44.2
	Compiler: Clang 16.0.0 with C++ ABI 1002
	KICAD_IPC_API=ON
```

chmorgan, 2025-03-29:
@gyohng tested here with latest plugin from 'main' and sorting seems to work as expected. Clicking on 'stock' first gives me entries with 0, then entries with the largest stock, same for mfg and description.

I can't reproduce.

gyohng, 2025-03-29:
Seems like fixed in b67f6b1cf79ea510e5cdaa560e9198e9025aaf03 a few hours after I submitted the issue. Thanks for checking.

closed 2025-03-29 by gyohng