Bouni/kicad-jlcpcb-tools #596: kicad9 - AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetPlotInvisibleText'

CarlFK, 2025-04-01:
plugin version 2025.3.1

I am happy to clone/install something and test.

I doubt the missing parts are relevant, but included for completeness. 

my project 
https://github.com/CarlFK/pimo/tree/main/project

```
carl@x1:~/src/pimo/project$ kicad pimo.kicad_pro 
DEBUG:com_github_bouni_kicad-jlcpcb-tools.store:Part J3 does not exist in the database and will be created from the board.
...
2025.04.01 15:54:18 - DEBUG - update_from_board -  Part C1 does not exist in the database and will be created from the board.

Traceback (most recent call last):
  File "/home/carl/.local/share/kicad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 870, in generate_fabrication_data
    self.fabrication.generate_geber(layer_count)
  File "/home/carl/.local/share/kicad/9.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/fabrication.py", line 136, in generate_geber
    popt.SetPlotInvisibleText(False)
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetPlotInvisibleText'
```

```
Application: KiCad x86_64 on x86_64

Version: 9.0.1-9.0.1-0~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.18

Platform: Ubuntu 22.04.5 LTS, 64 bit, Little endian, wxGTK, X11, ubuntu, wayland
OpenGL: Intel, Mesa Intel(R) HD Graphics 620 (KBL GT2), 4.6 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3

Build Info:
	Date: Mar 30 2025 15:49:29
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.1
	Curl: 7.81.0
	ngspice: 43
	Compiler: GCC 11.4.0 with C++ ABI 1016
	KICAD_IPC_API=ON

Locale: 
	Lang: en_US
	Enc: UTF-8
	Num: 1,234.5
	Encoded кΩ丈: D0BACEA9E4B888 (sys), D0BACEA9E4B888 (utf8)
```

chmorgan, 2025-04-02:
@CarlFK please try the version on 'main'. Should be all set.

chmorgan, 2025-04-02:
@CarlFK 2025.04.02 release has been made, can you try this and confirm it resolves your v9 issues?

CarlFK, 2025-04-02:
Yup.  worked.  thanks!  closing.

closed 2025-04-02 by CarlFK