Bouni/kicad-jlcpcb-tools #601: Plugin not found in 9.0.1 v2025.04.02 Linux

domw95, 2025-04-03:
**Describe the bug**
Plugin not appearing when refreshed on Ubuntu 24.10 Kicad 9.0.1. V2025.03.01 appears as a plugin, but doesnt generate files.

**To Reproduce**
1. Plugin cloned into $HOME/.local/share/kicad/9.0/scripting/plugins/kicad-jlcpcb-tools
2. checkout commit in repo
3. Refresh plugins in Kicad
4. main branch and 2025.04.02 don't appear,  V2025.03.01 does

Trace Output:
```
14:58:46: Trace: (KICAD_API) Manager: scanning system path (/usr/share/kicad/plugins) for plugins...
14:58:46: Trace: (KICAD_API) Manager: scanning PCM path (/home/dom/.local/share/kicad/9.0/3rdparty) for plugins...
14:58:46: Trace: (KICAD_API) Manager: scanning user path (/home/dom/.local/share/kicad/9.0/plugins) for plugins...
14:58:46: Trace: (KICAD_API) PCB frame: EDA_EVT_PLUGIN_AVAILABILITY_CHANGED
```

**Expected behavior**
Plugin found

# KiCad Version
Python 3.12, Ubuntu 24.10
```
Application: KiCad PCB Editor x86_64 on x86_64

Version: 9.0.1-9.0.1-0~ubuntu24.10.1, release build

Libraries:
	wxWidgets 3.2.5
	FreeType 2.13.3
	HarfBuzz 9.0.0
	FontConfig 2.15.0
	libcurl/8.9.1 OpenSSL/3.3.1 zlib/1.3.1 brotli/1.1.0 zstd/1.5.6 libidn2/2.3.7 libpsl/0.21.2 libssh2/1.11.0 nghttp2/1.62.1 librtmp/2.3 OpenLDAP/2.6.8

Platform: Ubuntu 24.10, 64 bit, Little endian, wxGTK, X11, ubuntu, wayland
OpenGL: Intel, Mesa Intel(R) Xe Graphics (TGL GT2), 4.6 (Compatibility Profile) Mesa 24.2.8-1ubuntu1~24.10.1

Build Info:
	Date: Mar 30 2025 15:48:28
	wxWidgets: 3.2.5 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.83.0
	OCC: 7.8.1
	Curl: 8.9.1
	ngspice: 43
	Compiler: GCC 14.2.0 with C++ ABI 1019
	KICAD_IPC_API=ON

Locale: 
	Lang: en_US
	Enc: UTF-8
	Num: 1,234.5
	Encoded кΩ丈: D0BACEA9E4B888 (sys), D0BACEA9E4B888 (utf8)

```

wburchard, 2025-04-15:
I've run into the same issue.  I was able to track this down to module distutils having been deprecated:

chmorgan, 2025-04-16:
Should be fixed with #611.

wburchard, 2025-08-08:
> @Bouni <https://github.com/Bouni> could you cut a release for this?

Any idea when it will show up in a release?  The last releaser is dated
April 2 of this year.

PR #611 (2025-06-13): Cmm distutils fix
Fixes #601. Even added some tests!