Bouni/kicad-jlcpcb-tools #602: Different Rotation displayed on the window vs in the exported CPL file

debreczeni, 2025-04-03:
**Describe the bug**
In the case described below the UI window displays a different value for the rotation compared to what it is exporting. 

My original issue was that the rotation regexes didn't work because of their order dependency until I noticed that the CSV export seems to be OK when overriding by component name, but the main UI window seems to be displaying a different value so the regexes are matched in a different order or with a different logic.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup an override for a rotation by name that would otherwise get rotated by a footprint match (a particular naming or regex order might be required to reproduce this - In my case the footprint match was the rule for `SOT-23` matching `SOT-23-TH-16`, and overridden by name rule for `TL074HIDYYR` as overriding with the footprint didn't do anything.
2. Click on Save and Update in the rotations window
3. Exit rotations window
4. Generate export, and cross-check the different rotation values on the main window and the generated CSV

**Expected behavior**
Rotations should match normally - they are different in my case as can be seen on the screenshots 

Being able to define a regex matching order could be a plus - but this should get its own issue.

[image]

[image]

[image]

# JLCPCB Toolkit Version
Most recent version, as of today: `2025-04-02`
It had the same behaviour on  version `2024-11-01` too.

# KiCad Version

```
Application: KiCad PCB Editor arm64 on arm64
Version: 8.0.9, release build
Libraries:
    wxWidgets 3.2.6
    FreeType 2.13.3
    HarfBuzz 10.2.0
    FontConfig 2.15.0
    libcurl/8.4.0 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.51.0
Platform: macOS Ventura Version 13.6.4 (Build 22G513), 64 bit, Little endian, wxMac
OpenGL: Apple, Apple M1 Max, 2.1 Metal - 83.1
Build Info:
    Date: Feb 18 2025 16:12:55
    wxWidgets: 3.2.6 (wchar_t,wx containers)
    Boost: 1.87.0
    OCC: 7.8.1
    Curl: 8.7.1
    ngspice: 44.2
    Compiler: Clang 16.0.0 with C++ ABI 1002
Build settings:
```

dhoepfl, 2025-04-09:
> [@dhoepfl](https://github.com/dhoepfl) is this in your wheelhouse now? ;-)

I’ll have a look.
I think this has to do with clockwise vs. counter-clockwise specification of the angle.

dhoepfl, 2025-04-11:
Not a problem of rotation direction:

The main window uses reference, then footprint to match the rotation to use.
The fabrication uses value, then footprint to match the rotation.

My suggestion would be to change it as follows:

* Unify it (obviously).
* Match in the following order: reference, value, footprint.
Matching by reference might especially makes sense once board local corrections are added: You might have two components with the same value/footprint but different rotations (e.g. C, 10uF, 6.3V vs. 25V).
* Add a column or add to the correction column an indication what matched: “(ref)”, ”(val)”, “(fpt)“

dhoepfl, 2025-04-11:
Related: #477.