Bouni/kicad-jlcpcb-tools #604: Right icon bar not wide enough / text is cut off

Schildkroet, 2025-04-08:
**Describe the bug**
Text on the right side of the panel is cut off / not fully visible. Resizing window does not help.

[image]

**To Reproduce**
Open JLCPCB Tools

**Expected behavior**
Text is fully visible

# KiCad Version

```
Application: KiCad PCB Editor x86_64 on x86_64

Version: 9.0.1, release build

Libraries:
	wxWidgets 3.2.7
	FreeType 2.13.3
	HarfBuzz 9.0.0
	FontConfig 2.15.0
	libcurl/8.11.1 OpenSSL/3.3.3 zlib/1.3.1 libidn2/2.3.7 libpsl/0.21.5 nghttp2/1.65.0

Platform: Freedesktop SDK 24.08 (Flatpak runtime), 64 bit, Little endian, wxGTK, X11, gnome, wayland
OpenGL: AMD, AMD Radeon RX 6650 XT (radeonsi, navi23, LLVM 19.1.7, DRM 3.59, 6.11.0-21-generic), 4.6 (Compatibility Profile) Mesa 24.3.4 (git-769e51468b)

Build Info:
	Date: Mar 30 2025 06:27:59
	wxWidgets: 3.2.7 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.87.0
	OCC: 7.9.0
	Curl: 8.11.1
	ngspice: 44.2
	Compiler: GCC 14.2.0 with C++ ABI 1019
	KICAD_IPC_API=ON

Locale: 
	Lang: de_DE
	Enc: UTF-8
	Num: 1.234,5
	Encoded кΩ丈: D0BACEA9E4B888 (sys), D0BACEA9E4B888 (utf8)

```

Bouni, 2025-06-10:
Are you using a HighRes Display? 

If so, what resolution?

Bouni, 2025-06-10:
So then its amybe a Linux issue. On my Windows Machine with the same resolution (but no 120Hz) it seems ok:

[image]

closed 2026-09-06 by z2amiller