Bouni/kicad-jlcpcb-tools #609: Duplicate entries in the corrections list

dhoepfl, 2025-04-11:
**Describe the bug**
Saving a correction entry with an regex that already exists in the list of corrections results in duplicate entries in the list.
Pressing “Delete” deletes all entries.
(The database is correct, only the latest entry is present.)

**To Reproduce**
1. Open the corrections dialog
2. Enter a new pattern, e.g. “TEST”
3. Click “Save”
4. Click “Save” again
5. Click “Delete”

**Expected behavior**
Saving a “new” entry with an regex that already exists should update the entry, even when nothing is selected.

Edge case: Entry “a” is selected, its regex is edited to “b”, which *also* exists. What should saving do? Delete “a” and update “b”?

# KiCad Version
Application: KiCad x86_64 on x86_64

Version: 9.0.0, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.2.0
	FontConfig 2.15.0
	libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.64.0

Platform: macOS Sequoia Version 15.4 (Build 24E248), 64 bit, Little endian, wxMac

Build Info:
	Date: Feb 19 2025 14:48:01
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.87.0
	OCC: 7.8.1
	Curl: 8.7.1
	ngspice: 44.2
	Compiler: Clang 16.0.0 with C++ ABI 1002
	KICAD_IPC_API=ON

Bouni, 2025-06-10:
I hope you'll submit a PR for this 😏

dhoepfl, 2025-06-10:
> I hope you'll submit a PR for this 😏

Might take a few days.
What do you think about the edge case I described? (Alert: Do (not) replace?)